arduino/Arduino #4265: IDE 1.6.6 breaks sketches that use multiline function declarations.

bperrybap, 2015-12-06:
The latest 1.6.6 IDE breaks sketches that use to and still should compile.
The issue is that the IDE is incorrectly inserting its prototypes when it converts the sketch to a .cpp file.
It appears that it is looking for first line of the first function and then attempting to insert the prototypes on the line just above it.
When the function declaration is not all on a single line, because the return type is on a line above the reset of the declaration, the IDE inserts the sketch function prototypes below the return type of the function declaration which causes the code to no longer compile.
This has broken many of my sketches and has broken some of my openGLCD library example sketches.
While the code in my sketches and library are much more complex and do not involved the setup() or loop() functions, here is a minimal example that demonstrates the problem.
```
void
setup(){}

void
loop() {}
```

If you go look at the .cpp file generated, you can see the problem:
```
#include <Arduino.h>
#line 1
#line 1 "/media/UbuntuRoot/home/bill/Arduino/Arduino-AVR/sketches/test/Test166/Test166.ino"

void
void setup();
void loop();
#line 4
setup() { }

void
loop() {}
```

Notice that the prototypes were inserted below the return type of the first function.
Curiously, it does manage to locate the and generate the correct prototype even if there are multiple whitespace lines between the return type and the reset of the function declaration or even if the rest of the declaration is on multiple lines like say the function name, parens, or arguments are on different/multiple lines.
The issue seems to be it is inserting the prototypes just above the line that contains the function name rather than just above the line that contains the return type.

ffissore, 2015-12-07:
Sorry, that's a known limitation, caused by ctags, a tool used by arduino-builder. Until @cmaglie comes up with a clang based replacement, please write your functions on one line only

ffissore, 2015-12-07:
This issue was moved to arduino/arduino-builder#80

closed 2015-12-07 by ffissore