arduino/Arduino #4266: 1.6.6 IDE is again issuing warnings for .git sub directories

bperrybap, 2015-12-06:
If you have a git repository inside a git repository the IDE will issue this warning:
> WARNING: Spurious .git folder in 'openGLCD' library

I believe this issue came up before in a previous release a while back and was agreed to resolve it by simply stopping issuing the warning since there are legitimate reasons for having a git repository inside a git repository.

per1234, 2015-12-07:
Try the [hourly build](https://www.arduino.cc/en/Main/Software#hourly). This was [fixed](https://github.com/arduino/arduino-builder/commit/f5edd23c01f0f21063b5000368cafff0e5cf4908) 2 weeks ago. Duplicate of https://github.com/arduino/Arduino/issues/4204.

ffissore, 2015-12-07:
Indeed fixed. It was a regression. Fix will be part of 1.6.7

closed 2015-12-07 by ffissore