arduino/Arduino #4269: indexOf() returning wrong value i compound statement

richw8341, 2015-12-07:
The expression       if(n=indexOf.someString("text")>=0)..........
will not return the correct index of "text"

This expression does work

                          n=indexOf.someString("text");
                          if(n>=0)............

VogonJeltz, 2015-12-07:
as strstr is a standard C lib function I would suggest to try the following as a workaround (and the syntax is IMO a little more C- compliant...):

~~~
int16_t  strstrpos(char * haystack,  char * needle)   // find 1st occurance of substr in str
{
   char *p = strstr(haystack, needle);
   if (p) return p - haystack;
   return -1;   // Not found = -1.
}
~~~

matthijskooijman, 2015-12-07:
The problem here is operator precedence: The `>=` operator is evaluated before the `=` assignment operator, so the result of the comparison is assigned to n. See also http://en.cppreference.com/w/cpp/language/operator_precedence

To fix this, you can add extra parenthesis: `if((n=indexOf.someString("text"))>=0)`

Since this is really a problem with your sketch, not the Arduino IDE, I'm closing this issue as invalid. If you have suggestions on where the documentation could have been better to prevent this isssue, feel free to still leave a comment.

richw8341, 2015-12-07:
My bad. Sorry. Even after thirty years I still cannot see my errors sometimes 

Thanks for that explanation..

closed 2015-12-21 by cmaglie