arduino/Arduino #4270: Bad permissions on Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude_bin

jhowe-sgs, 2015-12-07:
Using 1.6.6, the zipped file installs this file sans the execute bit. 

Installs 0644, should be 0755.

bertsierra, 2016-01-04:
I noticed that the same issue persists in Arduino IDE 1.6.7, released in December 2015.  Same issue: avrdude_bin executable ships without the execute bits enabled under OS X at least:

Flags before:
-rw-r--r--@ 1 admin  staff   369784 Apr 14  2015 avrdude_bin

Fix:  chmod 755 avrdude_bin  ## in admin account

Flags after:
-rwxr-xr-x@ 1 admin  staff   369784 Apr 14  2015 avrdude_bin

facchinm, 2016-01-04:
`avrdude_bin` is shipped without executable bit to avoid Gatekeeper warning on Yosemite and newer OSX (https://github.com/arduino/Arduino/issues/3468)
If you take a look at `avrdude` wrapper script (which gets called by the IDE) it takes care of adding the executable flag.

closed 2016-01-04 by facchinm

jhowe-sgs, 2016-01-28:
See #4501 for issues on a multi-user laptop deployment scenario.

Thanks