arduino/Arduino #4271: excessive wakeups el Capitan

Breidenbach, 2015-12-07:
system is complaining about excessive wakeups
Hourly build 12/4 1.6.7

[Arduino_2015-12-07-172304_Hal-Breidenbachs-MacBook-Pro.wakeups_resource.pdf](https://github.com/arduino/Arduino/files/54697/Arduino_2015-12-07-172304_Hal-Breidenbachs-MacBook-Pro.wakeups_resource.pdf)

ffissore, 2015-12-09:
I'm not a mac expert. Can this be related to #3210 ? If so, as soon as Oracle releases the fix and we bundle the new JVM, it will be fixed

gvarisco, 2018-12-11:
I'm not able to reproduce this in macOS High Sierra or Mojave. @Breidenbach I'm going ahead and mark this issue as 'closed', but feel free to re-open it if the problem persists in newer OS versions.

closed 2018-12-11 by gvarisco