arduino/Arduino #4273: Version 1.6.6: Bad permissions on Arduino.app/Contents/Java/reference/Edison_help_files, Arduino.app/Contents/Java/reference/Galileo_help_files

jhowe-sgs, 2015-12-08:
serveradmin@wenchang:/Applications/Arduino.app/Contents/Java/reference
$ ls -al
drwx------ 32 serveradmin staff 1088 Mar 25  2015 Edison_help_files
drwx------ 40 serveradmin staff 1360 Mar 25  2015 Galileo_help_files

Top level help dirs are mode 0700, should be 0755.

Thanks

NicoHood, 2016-08-24:
Another user reported the same error for (arch)linux. It applies to all files in the listed folders above too.

NicoHood, 2016-08-28:
Those files are downloaded here and the zip file already contains this data:
https://github.com/arduino/Arduino/blob/master/build/build.xml#L188

We have two options to fix this issue:
* Change the zip file (and its shasum)
* Merge this PR: #5306

PR #5306 (2016-08-29): Fixed reference folder permission on unix #4273
Fixes #4273
closed 2016-08-29 by cmaglie