arduino/Arduino #4275: Arduino Version 1.6.6 Serious install Problems!

RichW8VKinSidneyOhio, 2015-12-08:
I have installed Arduino Version 1.6.6 into a HP Windows 7 64 bit AMD machine with a Comm Port card set to coms 3&4. Not messing with my coom port card! No way! Problem is Arduino's Windows Installer puts this program in and it automatically installs it to comm ports either 3 or 4 and doesn't even see a USB ports on the computer. This is verified under the tools menu! This tools menu when going down to the Port should let you select a USB or Comm Ports. Never even shows a USB port choice! Fix the port choices as all it lets me choose from is comm 3 & 4! No USB Port Choices! I expect Help ASAP at my email address. This is a bug in your software! Can't do a thing until it's fixed! If it isn't fixed then I'm going to start raising Hell!

matthijskooijman, 2015-12-08:
The Arduino IDE shows all the serial ports it has found. This normally includes normal serial ports, and also any USB serial ports that are available. If the IDE does not show the USB serial ports, it is likely that the drivers for those are not installed, or not properly installed. You could check out the Windows device manager to see if the Arduino is listed there, and if it registers as a serial port.

Also, I would suggest reviewing your tone in subsequent comments. Staying polite and friendly usually invites more help than lots of exclamation marks and a demanding tone.

RichW8VKinSidneyOhio, 2015-12-08:
The USB stuff is working properly! It installes other devices such as a Garmin Nuvi 40 each time it is plugged in! The device manager never, I say never, shows the wrongly installed device as the arduino name when plugged in and out! I've even tried rear USB ports on this machine! Thanks for your answer, but I'd like to see you fixed this problem! Wow what a goofy install occurring! Thanks!

RichW8VKinSidneyOhio, 2015-12-08:
When looking in the Device Manager it is always listed as an unknown device! The tabs show nothing under properties. When plugging the 2560 in and out several times it always notes it's not installed correctly and not useable! The IDE after deleting the device driver and un installing the software doesn't re install the program right! The tools menu still selects comm port 3 &4. But when manually i go and run the dpinst-amd64.exe, it installs and says everything is ready! Wild isn't it! Going to uninstall and let computer set while. I've got to take a dog nap for several hours and re try things later again! I welcome anymore ideas you may have thru email! Thanks for being there and supporting!

matthijskooijman, 2015-12-08:
> When looking in the Device Manager it is always listed as an unknown device! 

That suggests that indeed the Arduino drivers are not installed properly.

> But when manually i go and run the dpinst-amd64.exe, it installs and says everything is ready! 

I'm not sure off-hand what that .exe is, but I suspect it might be the driver installer. But after you run it, it still lists the Arduino as an unknown device?

RichW8VKinSidneyOhio, 2015-12-08:
That's correct! In other words if I look what driver installed with windows 7 device manager it doesn't list it as an Arduino, it says unknown device! So the driver for a 64 bit machine doesn't work for this version of windows 7! And it seems like the next version of the IDE Software they need to do something under tools so you can slide a bar and try to force different comm ports up to 50 and also select a USB port selection! But they know more then me! And I have 2 brand new 2560s that I can't do a thing with! Other than waste time trying other drivers for other things and hope I might get lucky and one works! Otherwise you got any ideas? Thanks!

matthijskooijman, 2015-12-08:
> And it seems like the next version of the IDE Software they need to do something under tools so you can slide a bar and try to force different comm ports up to 50 and also select a USB port selection! 

That won't help - there is no comm port for your USB device. This needs a driver to tell Windows how to talk to the Arduino and how to translate serial messages into USB messages that the Arduino understands. The only way to fix this is to fix the driver installation.

Perhaps you can try manually installing the driver? IIRC that is through the device manager, double click the unknown device and then there should be some install driver or update driver button. I'm not entirely proficient with Windows, so this is pretty much where my expertise ends :-) You could try asking on the Arduino forums (http://forum.arduino.cc) for any other suggestions perhaps?

One thing that crossed my mind, is that perhaps your Arduino boards are manufactured by the competition at arduino.org and have a product identifier unknown to the arduino.cc driver. Could you see if the labels on the board say "arduino.cc" or "arduino.org"? And can you look up the USB VIDPID identifiers? How to do this is described here: http://www.makeuseof.com/tag/how-to-find-unknown-device-drivers-by-their-vendor-device-id/ (the first section under "The Manual Way from the Device Manager", the rest of the page is not relevant).

RichW8VKinSidneyOhio, 2015-12-08:
Only information in device Manager for this unknown device is under the driver tab! This is all I can tell you: Driver Version 6.1.7601.18328  06/21/2006  Windows didn't install the driver and reports error code 43! Thanks! Bought this off ebay! I'll return it and buy one off this web site! Absolutely no id on the board and is made in china is blue colored! That's it

matthijskooijman, 2015-12-08:
If you bought this off ebay, I suspect this is a clone that uses a CH340 serial chip, which is not supported by the drivers shipped with the Arduino IDE. You should contact the seller for information on what drivers to use, or perhaps look around on Google if you can figure out what serial chip is used. If you find the right drivers and get them working, it's likely that the board will work fine using the Arduino IDE from that point. Returning them and buying official Arduino boards is of course also a good option :-)

I'm closing this issue, since third-party hardware is not something that Arduino can support, nor is the goal of this issue tracker.

closed 2015-12-08 by matthijskooijman

RichW8VKinSidneyOhio, 2015-12-08:
I did what Microsoft said about error code 43! Uninstall, then scan for hardware changes it reinstalled driver then under properties General tab error code 28 and Details Tab got driver device is ACPI\AWY001. Thanks!