arduino/Arduino #4282: Wrong sketch lines numbering and nonexisting errors

sasa999, 2015-12-09:
#define MODE   1

#if MODE == 0
#include <Servo.h>
Servo myservo;  
#endif

//try compile this. With MODE = 0 it compiles without problem, but with MODE=1 it show non existing errors on non existing sketch lines

matthijskooijman, 2015-12-09:
I added ``` to your comment to mark the code area, so it won't mess up the less-than, greater-than :-)

Can you show the error you get? Also, could you try the latest hourly build from arduino.cc? Significant changes have been made, which I suspect might fix your problem.

sasa999, 2015-12-09:
[img]http://s23.postimg.org/l354kfhm3/scr3.jpg[/img]
[img]http://s7.postimg.org/ro4y67kpn/scr2.jpg[/img]
It show random non exixting errors. 
my sketches (all) worked fine till last night. with Arduino IDE 1.6.4 . Now not compiles all sketches vith this piece of code

matthijskooijman, 2015-12-10:
Did you try the latest hourly build version? That means the one in lower left of https://www.arduino.cc/en/Main/Software

You say this sketch worked fine with 1.6.4, but also with 1.6.5?

Can you also post the errors you get with the sketch you posted in the original message? You now posted a sketch without the errors, and errors without the sketch, which doesn't really help narrowing down the problem?

sasa999, 2015-12-10:
http://s16.postimg.org/asrx3nh1x/scr_blink.jpg
For example i can take just any code. 
Code is from Arduino examples - Blink. 
I added 
```
#define MODE  1

#if MODE == 0
#include <Servo.h>
Servo myservo;  
#endif
```
there is full code. Compiled for nano board, but on any board is errors
```
/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the Uno and
  Leonardo, it is attached to digital pin 13. If you're unsure what
  pin the on-board LED is connected to on your Arduino model, check
  the documentation at http://www.arduino.cc

  This example code is in the public domain.

  modified 8 May 2014
  by Scott Fitzgerald
 */
#define MODE  1

#if MODE == 0
#include <Servo.h>
Servo myservo;  
#endif

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin 13 as an output.
  pinMode(13, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
}
```

matthijskooijman, 2015-12-10:
In your screenshot, I see you are still using version 1.6.5, which is not the latest version. Could you please try the latest "hourly build" version (which is built from the latest bleeding edge version) to see if the issue hasn't been fixed in that version already? If you do not understand what I mean here (since I asked two times already), please also tell me, so I can explain in more detail.

sasa999, 2015-12-10:
Tnx for help, I will try hourly build at weekend. Does it fine for ESP8266? 1.6.6 was not good. I need version that is good for ESP8266!

matthijskooijman, 2015-12-11:
> Does it fine for ESP8266? 1.6.6 was not good at all.

I don't know about that, sorry (but I also do not recall any bug reports from the ESP maintainers about things to fix on the IDE side to fix ESP support, so hopefully things have been fixed on the ESP side).

sasa999, 2015-12-14:
Instelled Hourly Build with ESP8266 2.0.0
Board ESP-12e

Board nodemcuv2 (platform esp8266, package esp8266) is unknown
Error compiling.
.
changed board to Generic ESP8266

Board generic (platform esp8266, package esp8266) is unknown
Error compiling.

Totaly nothing!!!

sasa999, 2015-12-14:
http://s29.postimg.org/k1g2yyy13/scr8.jpg
What does it mean -  'actionSetup' was not declared in this scope? How fix it?
ActionSetup() is standalone function. It is writen next to loop(). Similar is Kill(), PageOne() etc. They all worked fine before updates.

matthijskooijman, 2015-12-14:
Can you tell us how you solved it?

Also, do these wrong line numberings when compiling for the ESP boards? Or do they also occur when you select e.g. the Arduino Uno? I don't think there is anything ESP-specific in the small example you posted? It might be better to test with the Uno, since that prevents any (unrelated) breakage related to the ESP core and newer versions for cluttering the debugging.

sasa999, 2015-12-15:
I removed ALL Arduino files (not libraries in My documents/Arduino) from comuter + 14.dec I download Hourly build  + esp8266 2.0.0. + deleted some my ESP8266 libraries (can't remember names, but they was shown as inavalid + deleted STM32 from hardware it was shown as invalid too, but worked fine with 1.6.4 + changed some variables, for example it didn't like int8_t reverse (without comments just show error compiation) - i changed to int8_t reverss  + moved ALL functions up , before loop(). + removed libc_replacements.c with my changes (in 1.6.4 it did not calculate right atan2 and I make some changes for fix it. In latest hourly buils atan2 is calculated right) + something was wrong with  time and I changed it to time2.. etc. ..  cant remember all changes :(
What exactly make wrong line numberings etc. - cant remember at 100% , but I think it was before I moved up my functions from bottom af page ,  variable names changing.
But very strange that with #define MODE  0 it compiles without problem (v.1.6.5), but for MODE 1 I need 1001 changes..

matthijskooijman, 2015-12-16:
> And question - HourlyBuild from 14.dec is the same 1.6.6 + some fixed bugs??

Yes, possibly new features too. The hourly build is made from the git master revision, you can see the changes in there using the "commits" tab here in github.

I finally had a bit of time to try your code for myself. I ran it in 1.6.5 (pasting the snippet from your first post above the bare minimum default sketch) and compiled for the Uno, and that broke. The errors were different, but investigating the modifed sketch source (which is actually fed to the compiler), I saw that the function declarations, and the associated `#line` directive, were inserted at the wrong place, between the `#if` and `#endif`. Since then the `#line` directive would be removed during actual compilation, line numbers would be wrong. Furthermore, the declarations would be removed, causing other errors.

I also tried using the latest nightly, and that compiles fine. The declarations now get inserted at the right place, because their generation was significantly improved in recent versions. Since we now know what the problem was, *and* it is working for you, I'll close this issue. Thanks for your feedback!

closed 2015-12-16 by matthijskooijman