arduino/Arduino #4283: gcc compiler failure after reinstall 1.6.5

VogonJeltz, 2015-12-09:
hello,
I tried to upgrade from 1.6.5 to 1.6.6 but nothing worked any more for my Due.
So I deleted 1.6.6 and reinstalled 1.6.5 from the zip for Windows.

Now I also can't compile any more - what's wrong ???

arm-none-eabi-gcc: error: C:\DOKUME~1\user1\LOKALE~1\Temp\build5037554308704768816.tmp/core/syscalls_sam3.c.o: No such file or directory
Fehler beim Kompilieren.

VogonJeltz, 2015-12-09:
WTF!
also the windows installer don't install correctly!!
from bareminimum.ino same error as above !!!

VogonJeltz, 2015-12-09:
update:
only a complete Ghost image reload could make the Arduino IDE compile again.

matthijskooijman, 2015-12-10:
I suspect you still had a newer SAM/Due core installed, which the (un)installer didn't remove. Downgrading it through the board manager might have helped. These cores are stored in ~/.arduino15/packages on Linux, not sure about Windows.

pierrotm777, 2015-12-10:
Delete all your files into the AppData\Roaming folder !

VogonJeltz, 2015-12-10:
that should happen automatically at a uninstall and/or a new install!

closed 2015-12-14 by VogonJ