arduino/Arduino #4285: Registry checked before environment variable

cosmiccube, 2015-12-10:
The issue is that the registry is being checked prior to the environment variables. This wouldn’t usually be a problem except that when using windows mandatory profiles the user shell folders are not modified at logon, only the environment variables which windows uses.

ffissore, 2015-12-10:
I didn't catch that
Yes, we check windows registry values before environment variables: registry values, if present, contain a more precise information about your user profile.
What's the error you're reporting?

cosmiccube, 2015-12-13:
Our system is set up that uses Windows Mandatory Profiles and it uses environment variables rather than registry. We cannot execute any program using the arduino IDE. For us to be able to use the Arduino IDE we need it to check environment variables first rather than the registry.