arduino/Arduino #4286: 1.6.6 on OS X El Capitan fails to Open file on Open With

baldengineer, 2015-12-10:
With multiple versions of the IDE installed, right-click on an INO file and select "Open With..."

If 1.6.6 is selected, the IDE will launch with the "BareMinimum" sketch, but will not open the INO file.

Same method but select an earlier version (I tried 1.6.1) and file open correctly.

ffissore, 2015-12-10:
I think this is a duplicate of #4057 and already fixed with latest hourly build. Can you give it a try? http://www.arduino.cc/en/Main/Software#hourly

baldengineer, 2015-12-10:
Yup, sorry about that. Didn't look far enough through 4057 to see the issue, version, and operating system changed midway through.

The latest hourly build doesn't have the issue.

closed 2015-12-10 by baldengineer