arduino/Arduino #4291: Burning Bootloader to Arduino Zero

Pecacheu, 2015-12-11:
I can't burn the bootloader to the Arduino Zero using ArduinoISP *(Duemilanove = Programmer, Zero = Target)*.

Before you select the board, Arduino IDE says "**incorrect device signature**", which is normal, because you must set the board type to the target board, even when using a programmer. But once I set the board to "**Arduino Zero**" (Native or Programming port), the IDE gives an error about not selecting a programmer, but the programmer is still set to "**Arduino as ISP**". I can use my Duemilanove to burn any other Arduino boards just fine.

facchinm, 2015-12-11:
Hi @Pecacheu ,
the Arduino Zero, not being an AVR, cannot be reprogrammed using ArduinoISP. 
In fact, the EDBG chip you can find on the Programming port will take care of burning the bootloader if `Atmel EDBG` is selected in `Programmer` submenu. No need of an external programmer :smile: 
Instead you can reprogram the Duemilanove and any other AVR based board using the Zero with `ArduinoISP` sketch as long as you take care about the different voltage levels.

closed 2015-12-11 by facchinm

matthijskooijman, 2015-12-11:
ArduinoISP is an AVR programmer, but the Arduino Zero is an ARM/SAMD board, so that is not expected to work. However, the Zero has the EDBG on-board, which can function as a programmer. IIRC you can burn the bootloader by selecting Arduino Zero (programming port) as the board, and Atmel-EDBG as the programmer (and of course connect to the programming port of your Zero).

See also "Burn the Bootloader" on https://www.arduino.cc/en/Main/ArduinoBoardZero

matthijskooijman, 2015-12-11:
Heh, didn't see @facchinm's comment, since I left this issue open over breakfast :-p Good to hear we're in agreement :-)

Pecacheu, 2015-12-11:
Okay, thanks. I re-burned my Zero's bootloader, but I also have a board that uses the same Atmel chip but doesn't have the EDBG on it. Would you know of any (cheap) programmers I could get that would work with it? I see it's not on the AVRISP's supported chip list.

matthijskooijman, 2015-12-11:
I remember reading that the JTAGICE3 supports it (with recent firmwares), and that you can remove some resistors from the Arduino zero to allow using the EDBG on an external board too, but I really haven't tried that. Here's where I read that, might be helpful to you: http://forum.arduino.cc/index.php?topic=330227.0

rishi-electronut, 2016-11-25:
Hey, I would like to know the command to burn the bootloader to an arduino zero using Atmel ICE through Windows command prompt.
I have been trying various commands discussed [here](https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc) and they all seem to work fine. However, I am unable to find the command that would allow me to burn the bootloader

facchinm, 2016-11-25:
@rishi-electronut The correct commandline for burning the bootloader is derived from https://github.com/arduino/ArduinoCore-samd/blob/master/platform.txt#L147 . 
The easiest way to get it for your configuration is enabling "Verbose upload" and executing "burn bootloader". In this way you'll get the commandline which you can execute in a Windows shell.

Deltoz, 2018-05-14:
Hi all.
It is possible to do on MKRWAN 1300 ?
How I can rewrite the bootloader on this board?
Thanks

Deltoz, 2018-05-17:
@facchinm Perfect! Thanks.
But what I have to use to do that?
What I have to connect the RESET, SWDIO and SWCLK Pins?
I do not understand what I have to use and how connect it :)

facchinm, 2018-05-17:
If you have an Atmel ICE with the "octopus cable", the connection is
```
USB SIDE

  x      3
  4      0
  2      x
```
So you need SWDIO, SWCLK, GND and RESET

Deltoz, 2018-05-17:
no i do not have an Atmel ICE :(
As programmer I have a Willem TL866CS and a UPA-USB 1.3 (or other Arduino's)
But I don't know if I can use that... not sure...