arduino/Arduino #4295: Wire lib: the first address that can be used is 8 ?

VogonJeltz, 2015-12-12:
hello,
I jsut read in the Wire reference:
However the addresses from 0 to 7 are not used because are reserved so the first address that can be used is 8. 
https://www.arduino.cc/en/Reference/Wire

Is that true?

Why?

I have several devices with a i2c device addr = 2  - how can they be used then ?

matthijskooijman, 2015-12-12:
I think the I2c specification mandates this, though I'm not entirely sure where to find this latest spec. These pages have a bit more info and seem to be somewhat authoritative:

 http://i2c.info/i2c-bus-specification#7-bit-addressing
http://www.i2c-bus.org/addressing/

If you have a device using address 2, it seems it is in violation of the specification, I guess (not entirely unheard of).

Note that a question like this is probably better asked on the forums or mailing list, keeping this issue tracker more for actual bugs and feature suggestions. I'll leave it open for now, feel free to close it yourself, or if you think the documentation should be improved, suggest an improvement.

VogonJeltz, 2015-12-12:
yes, thanks for the hint - I supposed that was a bug either in the Wire documentation or in the Wire specification. Finally for a 8-bit dev addr perhaps that would mean that addresses from 0 to 15 are not accessable.

Actually the i2c sensors by 0x02 dev addr are from Lego Mindstorms (actually, all by Lego and all by Mindsensors) which is not well-known for applying to either standards admittedly.

I would like keep this topic opened a couple of days and in case there won't arise new aspects I'll close it then.

Thanks for your input!

closed 2015-12-13 by VogonJeltz