arduino/Arduino #4298: IDE 1.6.6 incompatible with Win 10 on MS Surface Pro and others. Airplane Mode required?

stevenmblock, 2015-12-12:
Arduino IDE 1.6.6 "hangs" completely for up to a minute or more when starting on Win 10 on a MS Surface Pro, before. It ALSO hangs for any file save, and file open, and on closing.  It hangs on running Tools, or when compiling or uploading a sketch.  This is intolerable.,  Following comments in earlier threads, I ran with the suggestion to turn off Bluetooth. However, that did not solve the problem -- it still hung! Furthermore, both the MS Wedge mouse and the MS Stylus pen run off Bluetooth, so de-activating Bluetooth hobbles my use of the machine.  I finally discovered that setting the Surface in full Airplane Mode (both Bluetooth and WiFi turned off) was a workaround, and it eliminated all the hang-ups.  However, you not only lose the mouse and pen input this way, but also any Internet connectivity! WHY CAN'T SOMEONE FIX THIS ISSUE, once and for all?!?  One or another version of this problem has been around for years.  Or perhaps someone more expert can suggest a more practical workaround than going into Airplane Mode?  I could really use some help, here, and I am not alone!

ffissore, 2015-12-14:
The issue which solution is to turn off bluetooth has been solved back in 2013 with release 1.5.6. You're the first windows user reporting this issue. Since you're the first one, it may be related to your Surface. Since you said you've also deactivated wifi, it may also be due to a sluggish network

stevenmblock, 2015-12-14:
Thank you, Federico, for getting back to me so quickly.  I can tell you that this issue on the Surface Pro 3 (Win 10) is not related to a sluggish network.  It happens BOTH on my home wifi (Comcast is the ISP), which is fairly peppy, AND also at Stanford University, where the wifi is incredibly fast (and Stanford is the ISP). I can also report that the problem does not occur on a Lenovo X220 laptop running WIn 7 on the same two wifi networks. So it's not the network. I am upgrading the Lenovo X220 to Win 10 today, to see if the problem re-appears on that machine, and will report back to you about that.  Do you have any ideas about what might be causing the problem with the MS Surface/Win 10?  Do any other users report similar problems?

stevenmblock, 2015-12-14:
Ciao Federico,
I have more information for you. I can report that a Lenovo X220 laptop running (a newly-installed) Win 10 does NOT produce the "hang" problem running Arduino IDE 1.6.6.  And it is not a wifi speed issue, since the Lenovo is using the same LAN as the Surface.  So it must be something specific to the Surface Pro 3 that makes it incompatible with IDE 1.6.6.  Do you have any clues about what that might be?  Or what I could try?  All I know is that the problem disappears when the Surface is placed in Airplane mode.

stevenmblock, 2015-12-14:
Federico,

PROBLEM SOLVED!  I just downloaded and installed the latest version of JAVA (Version 8 Update 66), and the issue disappeared!!  I'm delighted.  I can now run both WiFi and Bluetooth without briging the Arduino IDE to its knees.  You might want to pass on this result to any other Surface owners experiencing similar problems.  Thanks again for your attention to this problem.

ffissore, 2015-12-15:
That's funny: the IDE already bundles java, you **shouldn't** need to download anything else

matthijskooijman, 2015-12-15:
> That's funny: the IDE already bundles java, you shouldn't need to download anything else

Wasn't this only on Linux?

matthijskooijman, 2015-12-15:
If also on Windows, perhaps the JDK installer changes something that a bundled java (which isn't installed as such, I expect) doesn't do?

ffissore, 2015-12-15:
Ahh good question: @cmaglie takes care of the installer

cmaglie, 2015-12-15:
The installer just unzip the files and do some operations like installing the drivers or setting some registry key to register the "uninstaller", but this is normal housekeeping that is completely unrelated with java AFAIK.

@stevenmblock have you tried the installer or the zip file?

stevenmblock, 2015-12-24:
Merry Christmas!  And thank you, Federico, Cristian, and Matthijs, for you attention to this problem.  Sorry to take so long to get back to you; I have had to attend to other things for a few days.

Unfortunately, my joy in thinking that I'd found a solution to this problem (by installing JAVA) was short-lived! The problem re-surfaced immediately after my the first reboot of my machine, a MS Surface Pro 3.  There is DEFINITELY a problem between Java's/IDE's file access and the Wifi operation of the Surface.

Here is what I know now, to recap:

1.  The Arduino IDE takes MINUTES just to start up, to open or save any file, to run Tools, or even to close the program on my Surface Pro 3 (running Windows 10) whenever its WiFi is active. Placing the Surface in Airplane mode (which deactivates both Bluetooth and WiFi), or de-activating JUST the Wifi (and leaving the Bluetooth active), fixes this problem. I can still use my mouse (Bluetooth) and Suface pen (Bluetooth) so long as I turn off WiFi only, and not have a problem.

2. I can still get internet access without experiencing problems if I use hardwired ethernet (through a USB-to-ethernet adapter) instead of WiFi.  Also, I can program the Arduino over a COM port and run ethernet at the same time if I use a 4-way USB hub.  This works without any problems.

3. The IDE issue has nothing to do with my particular network configuration (which is at Stanford University), because I don't experience the IDE problem on a Lenovo X220 laptop communicating over the identical WiFi router, nor do I have a problem with a desktop machine.  Also, I experience the problem on my Surface Pro 3 at home as well as at work, on a different WiFi router that uses a different ISP.

4. The IDE problem on the Surface happens in IDE versions 1.6.5, 1.6.6, and 1.6.7 (all that I have tried).   I really want to stay with v1.6.6 on up, anyway, because I'm taking full advantage of the new Serial Plotter feature.  These IDEs were all installed using the Windows installer, by the way. 

5) The Surface Pro 3 seems to use the Marvell AVASTAR Wireless-AC Network Controller, according to Windows' System Devices. I tried inquiring running a driver update, but got back the message "The best driver for your device is already installed," so I guess there is nothing to change there.

Does anyone have any idea how to fix this problem, or what might be causing it??

stevenmblock, 2015-12-27:
I have a bit more to report:

I have Cisco AnyConnect Secure Mobility Client installed on my Surface Pro 3, to do VPN to my university.  When the VPN is connected (over WifFi), the Arduino IDE start/load/exit delays disappear! But the moment I disconnect the VPN, the problem re-appears.  Usually, problems like this are the other way around -- this one happens when I DON'T run the VPN!!

As an experiment, I tried these things:
1) Disabling the VPN client in Device Manager and rebooting.  Result:  IDE problem happens whebnever WiFi is connected.
2) Uninstalling the VPN client from my machine altogether and rebooting. Result: IDE problem happens whenever WiFi is connected.  

This one has be baffled.  I can make the IDE start/load/exit problem go away by either turning WiFi off altogether, OR I can get Wifi to coexist with the IDE as long as the VPN client software is running AND connected, but not when it's running and disconnected.

Any ideas, anyone?!?

facchinm, 2016-01-04:
Hi @stevenmblock , thanks for the great description of the problem you are experiencing. 
It might be related to `mdns` discovery of network boards (like Yun). IDE uses latest `jmdns` version (3.4.1) and a discovery is triggered every 500milliseconds, maybe something in your network topology makes this code misbehave. Have you tried running `arduino_debug.exe` and see if any error is reported? Can you past them here eventually?

stevenmblock, 2016-01-15:
I tried running arduino_debug.exe but it reports nothing. Alas.

stevenmblock, 2016-06-13:
I AM PLEASE TO REPORT THAT **THIS BUG IS FIXED** in Arduino ID v. 1.6.9!!!!  They changed the way COM ports were identified in Windows, and evidently this change fixed the issue.  So if you encounter it, simply download the latest version of the IDE.  Thanks to one an all!

closed 2016-06-13 by stevenmblock