arduino/Arduino #4300: Missing font in Linux causes IDE menus to all dissapear

kravlin, 2015-12-14:
I've been experiencing the same issue as [here](https://www.reddit.com/r/arduino/comments/3u1pxn/missing_arduino_ide_menus_xpost_rarchlinux/)

After some testing and asking around, the issue was fixed by installing microsoft's propietary font packages. I'm not sure exaclty which font package fixed the issue at this point. I did have several Monospaced fonts already installed, so I don't think that's the issue font.

No errors were shown in the console, regardless of the -debug flag.

I actually don't have enough information or knowledge of the IDE to debug it or I'd provide more exact information, I just figure someone should know.

ffissore, 2015-12-14:
Do you have the issue even when using the IDE provided by arduino? https://www.arduino.cc/en/Main/Software

kravlin, 2015-12-14:
Yep. I downloaded directly from the arduino website.

What i'm expecting is that there's not a font included in some family or something that isn't included in arch by default.

ffissore, 2015-12-14:
We don't use specific fonts. We use "monospace" and "serif", which are provided by the OS. So I don't know what we could do to fix this. However, I've just become aware that font antialiasing was also not working properly on Arch and I fixed that in 6c5e58445402597f0472fbf4da9e518ff38b399b

kravlin, 2015-12-16:
It's very possible that arch doesn't ship with serif. as i'm looking through my fonts on my local system and only seeing one serif font, which is installed as part of the liberation fonts package.

So it's actually very possible that arch isn't shipping with serif or they're not installed in the right place, which is the issue. I'll do more digging with a new install of arch over the next couple days.

kjav, 2016-05-21:
I had this problem too, which I fixed by installing the ttf-ms-fonts package from the AUR. It seems like the text in the menus of the IDE use one of the following fonts:
  - andale mono
  - arial
  - arial black
  - comic sans MA (probably not this one)
  - courier new
  - georgia
  - impact
  - times new roman
  - trebucet MS
  - verdana
  - webdings (probably not this one either)
[Source](https://aur.archlinux.org/packages/ttf-ms-fonts/)

kjav, 2016-05-21:
Perhaps the default JMenuBar font is an ms-font? Would setting the menubar's font [here](https://github.com/arduino/Arduino/blob/60b169fbe93c861b6c06d57f57b8f5b2be5919e0/app/src/processing/app/Editor.java#L516) to editor.font solve the issue?

chrstphrchvz, 2016-08-08:
I'm not sure JMenuBar requires an MS font, since I have used this on systems without one. The font that it appears to try using can depend on what the desktop environment (if any) has set, e.g. for MATE, it is whatever is specified for "Application font" in Appearance settings, i.e. stored as org.mate.interface font-name; similarly for GNOME in [org.gnome.desktop.interface](http://stackoverflow.com/a/11873232). What desktop environment/window manager/theme does the issue occur on, and what fonts do they specify?

I tested the Swing GTK LaF menu bar alone by compiling and running the [MenuLookDemo example](https://docs.oracle.com/javase/tutorial/uiswing/examples/components/MenuLookDemoProject/src/components/MenuLookDemo.java) with the command [`java -Dswing.defaultlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel components.MenuLookDemo`](https://docs.oracle.com/javase/tutorial/uiswing/lookandfeel/plaf.html#commandLine). This might be a simpler method for reproducing the issue. So I think this is maybe an upstream issue at least for either Swing GTK look and feel (which is [what Arduino is defaulting to] (https://github.com/arduino/Arduino/blob/master/build/linux/dist/arduino#L28)), or specific desktop/windowing software. I haven't worked out how to programmatically find the exact font Java is told to use, but some of the needed statements might be found somewhere in [com.sun.java.swing.plaf.gtk.GTKLookAndFeel](http://grepcode.com/file/repository.grepcode.com/java/root/jdk/openjdk/8u40-b25/com/sun/java/swing/plaf/gtk/GTKLookAndFeel.java#FontLazyValue).

chrstphrchvz, 2016-08-08:
Another workaround might be to edit the [arduino Linux shell script](https://github.com/arduino/Arduino/blob/master/build/linux/dist/arduino#L28) to start Arduino with a different look and feel, although I don't know if this breaks anything. For example, to use the Metal LaF:
```
# Collect options to java in an array, to properly handle whitespace in options
JAVA_OPTIONS=("-DAPP_DIR=$APPDIR" "-Dswing.defaultlaf=javax.swing.plaf.metal.MetalLookAndFeel")
```

symbioquine, 2021-01-11:
I have a similar issue that just started showing up under Arch Linux recently (without changing versions of Arduino IDE) - it seems to affect both 1.8.10 and a version of nightly I downloaded some time back...

[image]

For me, the work-around seems to be just deleting the whole `java` folder within the arduino-1.8.10 distribution and letting it use my OS' java install. Obviously YMMV :-P

symbioquine, 2021-01-11:
Possibly related: https://github.com/arduino/Arduino/issues/11150

wrightrocket, 2021-01-17:
I modified the PATH that is set in the arduino script to not update the path or search the ./java directory:
@@ -15,14 +15,9 @@ export CLASSPATH
 LD_LIBRARY_PATH=$APPDIR/lib${LD_LIBRARY_PATH:+:$LD_LIBRARY_PATH}
 export LD_LIBRARY_PATH
 
-export PATH="${APPDIR}/java/bin:${PATH}"
-
 export JAVA_TOOL_OPTIONS=`echo $JAVA_TOOL_OPTIONS | sed 's|-javaagent:/usr/share/java/jayatanaag.jar||g'`
 
 JAVA=java
-if [ -x "$APPDIR/java/bin/java" ]; then
-  JAVA=$APPDIR/java/bin/java
-fi
 
 # Collect options to java in an array, to properly handle whitespace in options
 JAVA_OPTIONS=("-DAPP_DIR=$APPDIR")

wrightrocket, 2021-01-17:
1. Install the fonts from https://downloads.sourceforge.net/project/mscorefonts2/rpms/msttcore-fonts-installer-2.6-1.noarch.rpm 
2. Delete the following lines in the arduino script:

export PATH="${APPDIR}/java/bin:${PATH}"

if [ -x "$APPDIR/java/bin/java" ]; then
  JAVA=$APPDIR/java/bin/java
fi