arduino/Arduino #4301: Due: issues about Wire plus SPI, DueTimer + Scheduler

VogonJeltz, 2015-12-14:
is it possible that there are still unresolved issues about Wire plus SPI, DueTimer + Scheduler ?
I'm always runing into issues for SD init (either if SD.h or SdFat) whe running those different other libs additionally - when outcommenting SD init, everything runsd fine.

see, e.g.
http://forum.arduino.cc/index.php?topic=365405.msg2520360#msg2520360

VogonJeltz, 2015-12-14:
the whole pin setup is
```
// pins   0, 1       : Serial
// pins   2, 3       : ...
// SPI_SS 4          : (reserve)
// pins   5, 6       : ...
// pins   7,8,9,10   : motor pwm
// pins   11 - 13    : ...
// pins   22 - 37    : motor dir + encoder
// pins   38 - 46    : keypad pins
// pins   47 - 50    : bumper
// TFT_CS      51
// TFT_DC      52
```

the setup procedure is the following:
```

void setup()
{
  int32_t i;
  char    sbuf[128];

   // SD_CS       53
   // TFT_CS      52
   // TFT_DC      51
   
   
   //------------------------------------------------------------------------------------
   // Serial terminal window
   
      i=115200;
      Serial.begin(i);
      sprintf(sbuf, "Serial started, baud=%ld \n");
      Serial.println(sbuf);
      // lcdprint(sbuf);     
      Serial.println("[done.] \n");

 
   //------------------------------------------------------------------------------------
   // TFT LCD
   
      Serial.println();
      LCDTYPE = _ILI9341due_;
      Serial.print("init LCD... \n");     
      initLCD(1);   
      lcdcls();
      sprintf(sbuf, "LCD=%d wi%dxhi%d Font %dx%d \n",LCDTYPE,LCDmaxX,LCDmaxY,fontwi,fonthi);
      Serial.println(sbuf);
      Serial.println();
      lcdcls(); lcdprint(sbuf);
      Serial.println("[done.] \n");   

   //------------------------------------------------------------------------------------
   // init i2c devices
   
      sprintf(sbuf, "i2c init \n");
      lcdprint(sbuf);
      Wire.begin();       
      // i2c CMPS11: addr=0x60
      sprintf(sbuf, "CMPS11 Example V: %d \n", IMU_soft_ver()); // software version of the CMPS11
      Serial.println(sbuf);
      lcdprint(sbuf);
      Serial.println("[done.] \n");

 
 
   //------------------------------------------------------------------------------------
   // motor settings
   // setup for L293D motor driver
[... 146 lines cut ...]
// SD init
//=====================================================================================

int16_t initSD() {
   char sbuf[128];
   uint32_t  tstamp;
   int16_t   ior=0;
 
   tstamp = clock();
   ior = SD.begin(sd_cs);  // success==1==true; else 0==false
   while( !ior) {     
      sprintf(sbuf, "#: ...SD not found... ");
      curlf(); lcdprint("#: ...SD not found... ");
      Serial.println(sbuf);
      delay(1000);   
      ior=SD.begin(sd_cs);
      if (clock()-tstamp>20000) {Serial.println("#: ...break!"); break; }
   }
  if(!ior) return fileIO_ERR_SDCARD;     // SD ioresult==0 => error = -16
  return fileIO_OK ;                     // SD ioresult==1 => ok = 1
}

```

// SD_CS       53

peabo7, 2015-12-14:
How many times do you have to be told that this is the wrong place to post non-IDE issues?

Now you have copy-pasted someone else's non-IDE issue here.

Reading your messages is an astounding waste of time.

VogonJeltz, 2015-12-14:
no, they are my own (aka ArthurD) issues, and it seems its caused by Arduino IDE and lib and timer incomatabilities.

closed 2017-02-08 by agdl