arduino/Arduino #4302: Issue with I2C lcd library in IDE 1.6.6 and 1.6.7

RoRut, 2015-12-14:
I am running Windows10 ( 64 bit). When trying to use the I2C lcd library in IDE 1.6.6 and 1.6.7 it only prints the first character of each line on the display.

If I load IDE 1.6.5 everything works fine.

Please advise ??

Best regards,
Roland Rutsaert

facchinm, 2015-12-14:
Hi @RoRut ,
the problem is not within the IDE itself but it's caused by a misbehaviour of LiquidCrystal_I2C library (read here for a fix https://github.com/arduino/Arduino/issues/4142#issuecomment-156690585)
In fact the latest Wire needs to know how many characters are written and the LiquidCrystal library was reporting 0, thus stopping after the first character.
In fact the fix was merged but the library was not released, so you need to apply the fix manually.

cmaglie, 2015-12-14:
@RoRut 
please comment also here https://github.com/arduino/Arduino/issues/4142#issuecomment-156690585 and ask the author of the library to make a new release.

cmaglie, 2015-12-14:
ehm.. sorry I mean here: https://github.com/marcoschwartz/LiquidCrystal_I2C/pull/5

closed 2015-12-14 by cmaglie

RoRut, 2015-12-15:
> ehm.. sorry I mean here: marcoschwartz/LiquidCrystal_I2C#5 <https://github.com/marcoschwartz/LiquidCrystal_I2C/pull/5>

Did the mod  0 to 1.

 

Everything Works fine.

 

Thanks for the quick reaction.

 

Best regards,

 

Roland Rutsaert