arduino/Arduino #4303: 1.6.6 Library Manager - CRC errors occurred during updating libraries

Myriads, 2015-12-14:
During updating libraries, CRC errors are occurred:

- FreqCount,  MsTimer2, Time and TimeOne
- FreqCount and MsTimer2 updated manually after CRC errors occurred.

[image]

ffissore, 2015-12-14:
Please delete files from `%LOCALAPPDATA%\Arduino15\staging\libraries` and retry

Myriads, 2015-12-14:
Even if %LOCALAPPDATA%\Arduino15\staging\libraries were deleted, CRC errors still come out

Myriads, 2015-12-15:
In DownloadableContributionsDownloader.java, FileHash.hash function generates a quite different hash value with the checksum value of the Time library:

checksum : SHA-256:1b658b9d71cddd4c24a63ec5c0c1cf345a44b32d46e0a207c7e129f8bc6e6492
hash value : SHA-256:3ae1f8839a6d48aeb3c0c4489e0ddeb32b245df700b8fb01b03f784a47aee18e

Tested with Time library.

bukrat, 2016-04-28:
I have the same problem here. Any help?

Myriads, 2016-04-28:
Please try with the newest IDE version.

cmaglie, 2016-11-09:
This issue has been fixed with #5453.
The fixed version is available in the hourly build, or the next IDE release 1.6.13.

closed 2016-11-09 by cmaglie