arduino/Arduino #4305: Linux Gtk-WARNING

NicoHood, 2015-12-14:
I get this error whenever I startup my IDE (for month now) and I am sick of it now ;)
Its just a command line debug print, and it might be related to my OS.

`(java:4923): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",`

It would be nice to know why this happens and how to fix it. A better description in the warning could be an idea as well.

I am using Elementary Os.

NicoHood, 2015-12-14:
@matthijskooijman said I should close this issue since its a JDK bug and there is likely no chance to fix this inside the Arduino IDE.

closed 2015-12-14 by NicoHood

matthijskooijman, 2015-12-14:
yeah, GTK is known to spew warnings and failed assertions all over the place, so people usually ignore them. It's probably caused by something in the JDK or even GTK itself and unlikely to be fixable in java.