arduino/Arduino #4307: Liquid Crystal 1.0.4 is not loading properly or updating

a4jp-com, 2015-12-14:
Liquid Crystal 1.0.4 is not loading properly or updating. Liquid Crystal 1.0.4 loads 1.0.1

[image]
[image]

per1234, 2015-12-15:
@a4jp-com which Boards do you have installed? This sounds like the same issue described in https://github.com/arduino/Arduino/issues/4189 which I have found to be caused by libraries bundled with a boards package that have the same name as an Arduino IDE Built-In library.

ffissore, 2015-12-15:
Sounds like a duplicate of #3904 which has been fixed in latest hourly build http://www.arduino.cc/en/Main/Software#hourly

a4jp-com, 2015-12-15:
I had the hourly build 2015-12-14 11:21 and the problem was similar but 1.0.4 wouldn't load. So I'm not sure if this is a slightly different problem but I'll redownload the latest hourly build again now and see if the problem is still here. I just keep getting the "Update available for some of your libraries" popup.

a4jp-com, 2015-12-15:
I just updated to nightly build 2015/12/15 12:43 and the problem hasn't been fixed or this is a slightly different problem.

Library Manager: Type Updateable

Click More info for the Liquid Crystal by Arduino, Adafruit.

Select version 1.0.4.

Install.

Version 1.0.1 is shown.

Verson 1.0.4 is now missing but version 1.0.1 os loaded.

Select 1.0.3.

Install.

1.0.4 is in the list again.

per1234, 2015-12-17:
@a4jp-com it's clear that your issue isn't #3904. I pretty sure I know the cause and the solution but if you won't answer my question then I can't help you.

a4jp-com, 2015-12-17:
Sorry about that. I'm using a RAMBo board.

per1234, 2015-12-17:
Please do this:
- File > Examples > LiquidCrystal > Autoscroll
- Sketch > Show Sketch Folder
- Copy the path of the folder that opens when you do this and post it here.

a4jp-com, 2015-12-17:
I can check the file I loaded to get the board data as soon as I get home. It was from the Marlin firmware group.

a4jp-com, 2015-12-17:
This is where the board data comes from:
https://raw.githubusercontent.com/marlinfirmware/MarlinDev/platform/package_marlinfirmware_index.json

And here's the path:
C:\Users\design\Documents\Arduino\libraries\LiquidCrystal\examples\Autoscroll

per1234, 2015-12-17:
Yes the Marlin board package is the cause of the issue. Because the LiquidCrystal library bundled with the Marlin package has the same name as the LiquidCrystal library included with the Arduino IDE the Library Manager gets confused. I have submitted a pull request to the MarlinFirmware repository to fix this: https://github.com/MarlinFirmware/MarlinDev/pull/309. You can fix this problem on your system by opening the file
C:\Users\design\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.6.7\libraries\LiquidCrystal\library.properties
and change the line:
`name=LiquidCrystal`
to 
`name=LiquidCrystal(Marlin)`

a4jp-com, 2015-12-17:
Thank you so much @per1234 . That's fixed the updates ^^

closed 2015-12-17 by a4jp-com