arduino/Arduino #4308: NEW I-Bot library

somsinchai, 2015-12-15:
Please add my I-Bot library to the Arduino Library Manager.

https://github.com/somsinchai/IBot

thanks.

VogonJ, 2015-12-15:
could you provide a website with a project description, please?

somsinchai, 2015-12-15:
https://www.facebook.com/ibotcamp/?fref=ts
it about camp for hight school student to pratcit robotic skill our robot is a sample line follower robot and some servo using arduino UNO

cmaglie, 2015-12-16:
@somsinchai 
to publish the library you must create a "tag" or a github's "release", please have a closer look here:

https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager

somsinchai, 2015-12-16:
ok, now i've released my library.
thank.

cmaglie, 2015-12-17:
@somsinchai 
your library is named I-Bot but in the property file you wrote "WebServer", but maybe this is not what you want?
I'd say that you should write the correct name in library.properties, do another release and remove the previous wrong release.

somsinchai, 2015-12-17:
oh, my wrong
i missing edit library.properties form default file.
now i 've edit it and delete previous release.
thank

cmaglie, 2015-12-18:
Thanks, the library is now published!

closed 2015-12-18 by cmaglie