arduino/Arduino #4310: 1.6.6 - Board Manager - Loading issue

4dsystems, 2015-12-16:
There appears to be an issue with the Board Manager, where if you put a new JSON link in Preferences, then open up the Board Manager, it does not add it to the list. You have to close the Board Manager and then open it again, then it appears.

We did some testing and after adding the JSON, restarted the IDE, then opened the Board Manager, but same deal - the new listing didnt show up until we closed the Board Manager and then opened it again.

Just trying to write instructions on how to add a board, and found this quirk.

Thanks

lmihalkovic, 2015-12-16:
Just watched a video from adafruit a couple days ago about developping a library "... have to stop and restart the IDE when ...".

This non-refresh issue is simple to fix and creating a lot of un-necessary frictions in what is advertised as a "simple and convenient" tool. IMHO it should be a high priority to fix this issue. In my own branch I intend to eventually redo that code.

per1234, 2015-12-16:
This issue has already been reported(https://github.com/arduino/Arduino/issues/3795) and solved in the hourly build.

4dsystems, 2015-12-16:
Thank you for reference to issue 3795. Will document fix available in nightly build and 1.6.7 release, when that is available.

closed 2015-12-16 by 4dsystems