arduino/Arduino #4311: Serial plotter issue

pierrotm777, 2015-12-16:
[image]

We would see only five squares on the right top and not all this line of squares !

facchinm, 2015-12-16:
Hi @pierrotm777 , could you please paste here a sketch that produces that misbehaviour? Thanks

pierrotm777, 2015-12-16:
Sorry , my code was bad and the plotter run well but the 5 th square use the same color as the first !!!!
[image]

pierrotm777, 2015-12-16:
Is it possible to have the option to add text for each plot ?
[image]

pierrotm777, 2015-12-16:
To have the real value for each plot on the vertical axe would be cool !

facchinm, 2015-12-16:
I believe that the four colours limit and the missing labels are all known "issues" and @henningpohl  is on them.
Adding @agdl to the thread since he surely knows more

agdl, 2015-12-16:
Hi, no at the moment labels are not available. I opened an issue about this fact because il will be really cool to implement it

closed 2015-12-16 by agdl

pierrotm777, 2015-12-16:
I have found a the way for change number of different plot's color into the file theme.txt
plotting.graphcolor.size = 5
plotting.graphcolor.00 = #2c7bb6
plotting.graphcolor.01 = #fdae61
plotting.graphcolor.02 = #d7191c
plotting.graphcolor.03 = #abd9e9
plotting.graphcolor.04 = #fbd9e9

pierrotm777, 2015-12-16:
Some time , the console return me these errors

Exception in thread "AWT-EventQueue-0" java.util.NoSuchElementException
	at processing.app.helpers.CircularBuffer.min(CircularBuffer.java:42)
	at processing.app.SerialPlotter$GraphPanel.computeBounds(SerialPlotter.java:90)
	at processing.app.SerialPlotter$GraphPanel.paintComponent(SerialPlotter.java:119)
	at javax.swing.JComponent.paint(JComponent.java:1056)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JLayeredPane.paint(JLayeredPane.java:586)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paintToOffscreen(JComponent.java:5217)
	at javax.swing.BufferStrategyPaintManager.paint(BufferStrategyPaintManager.java:290)
	at javax.swing.RepaintManager.paint(RepaintManager.java:1272)
	at javax.swing.JComponent.paint(JComponent.java:1042)
	at java.awt.GraphicsCallback$PaintCallback.run(GraphicsCallback.java:39)
	at sun.awt.SunGraphicsCallback.runOneComponent(SunGraphicsCallback.java:79)
	at sun.awt.SunGraphicsCallback.runComponents(SunGraphicsCallback.java:116)
	at java.awt.Container.paint(Container.java:1975)
	at java.awt.Window.paint(Window.java:3912)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:842)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:814)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:814)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:789)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:738)
	at javax.swing.RepaintManager.access$1200(RepaintManager.java:64)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1732)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)