arduino/Arduino #4314: Arduino 1.6.7 doesn't open at all on Mac

GWBasic, 2015-12-18:
Arduino 1.6.7 doesn't open at all on Mac. When I try to open it, MacOS reports: the application "arduino" cannot be opened

An older version of Arduino, alpha 0023, opens fine.

PaulStoffregen, 2015-12-18:
Maybe you could try 1.6.6 (from November 2015) or 1.6.5 (October 2015) or 1.6.4 (May 2015) or other versions within the last year, rather than 0023 (from November 2011)?

You might also at the very least tell us which version of Mac OSX you have?

GWBasic, 2015-12-18:
I'm able to work around this problem by running this command in terminal:
/Applications/Arduino.app/Contents/MacOS/Arduino

I'm running the newest MacOS.

I suspect that some file attributes aren't preserved in the zip file; or that there is a code signature problem.

GWBasic, 2015-12-18:
1.6.6 doesn't work. 1.0.6 works.

cmaglie, 2015-12-23:
@GWBasic 
it's quite difficult to diagnose without further information.

May you try to open Console (`/Applications/Utilities/Console.app`) and after that try to open Arduino? I hope that the reason for the error is displayed there.

Can you also copy here your settings for "Settings -> Security & Privacy"?

roypardi, 2016-01-10:
I just downloaded v1.6.7 and have the same issue

**OS Verision:** OSX 10.11.12
**Security (Gatekeeper) settings:** "anywhere"
**Console log:**

```
failed to act on a ping it dequeued before timing out.
Jan 10 07:29:07 dazzle lsd[269]: LaunchServices: Could not store lsd-identifiers file at /private/var/db/lsd/com.apple.lsdschemes.plist
Jan 10 07:29:07 --- last message repeated 1 time ---
Jan 10 07:29:07 dazzle com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x1000003d.JavaApplicationStub[14471]): Could not find and/or execute program specified by service: 2: No such file or directory: /Applications/Arduino.app/Contents/MacOS/JavaApplicationStub
Jan 10 07:29:07 dazzle Finder[689]: spawn_via_launchd() failed, errno=111 label=[0x0-0x38d38d].cc.arduino.Arduino path=/Applications/Arduino.app/Contents/MacOS/JavaApplicationStub flags=1 : LaunchApplicationClient.cp #963 LaunchApplicationWithSpawnViaLaunchD() q=com.apple.root.default-qos
Jan 10 07:29:07 dazzle Finder[689]: spawn_via_launchd() failed, errno=111 label=[0x0-0x38d38d].cc.arduino.Arduino path=/Applications/Arduino.app/Contents/MacOS/JavaApplicationStub flags=1
Jan 10 07:29:07 dazzle lsd[269]: LaunchServices: Could not store lsd-identifiers file at /private/var/db/lsd/com.apple.lsdschemes.plist
```

cmaglie, 2016-01-11:
@roypardi 
I see from you log:

```
Could not find and/or execute program specified by service: 2: No such file or directory: /Applications/Arduino.app/Contents/MacOS/JavaApplicationStub
```

so for some reasone the file `/Applications/Arduino.app/Contents/MacOS/JavaApplicationStub` disappeared?

Well at least you solved it, even if we didn't find the cause (I see you edited your message to report that, BTW the next time make a new post so we get the notifications!)

@GWBasic for now I'm closing this, feel free to reopen when you have the time to provide further informations.

closed 2016-01-11 by cmaglie

richardloxley, 2016-03-07:
I have just had exactly the same symptoms - Arduino 1.6.7 will run from the desktop on OSX 10.11.3, but fails when copied to the Applications folder, with errors in the console log about not being able to find /Applications/Arduino.app/Contents/MacOS/JavaApplicationStub

I haven't been able to find the cause, but I've found a sequence of actions that fixes the problem.  If I dragged the app from the desktop to the root of the hard drive, I get a pop-up asking me to authenticate (which you don't get if you just drag it to /Applications).  After authentication, I then dragged the app from the root of the drive into /Applications.  At that point it now works.

I then guessed that it's some kind of permissions problem, presuming the normal copying to /Applications runs as some privileged 'installation' user, which perhaps doesn't set the permissions quite right?

But I then checked the resultant app directories, and the contents are identical, and the file permissions are the same.  The only difference was the one that doesn't work had all the files in the 'staff' group, and the one that does work had the files in the 'admin' group.  I tried doing "chgrp -R admin Arduino.app/" but that didn't make a difference.

I also looked at the extended attributes of both directories using "xattr -lr Arduino.app/" (no differences), and the access control lists and file flags using "ls -aleRO Arduino.app/" (also no differences).

All of which just tells us what isn't the problem, not what is, I'm afraid :-(

But at least there's a possible workaround (copy it into the root of the drive, then into Applications, as two steps), which might help anyone else with the problem.

Anyone know what extra steps OSX does when copying an app into the Applications folder that might influence this?

mikaeljorhult, 2016-04-08:
I just ran into the same problem on OS X 10.11.4. Trying to run Arduino.app after copying it to the Applications folder will result in an error saying that it can not be opened. This happens with both 1.6.8 as well as the latest hourly build of 1.6.9 even when Gatekeeper is deactivated.

Checking the console reveals a couple of errors occurring when trying to start the application:
```
LaunchServices: Could not store lsd-identifiers file at /private/var/db/lsd/com.apple.lsdschemes.plist
```
```
(com.apple.xpc.launchd.oneshot.0x10000026.JavaApplicationStub[11494]) Could not find and/or execute program specified by service: 2: No such file or directory: /Applications/Arduino.app/Contents/MacOS/JavaApplicationStub
```
```
spawn_via_launchd() failed, errno=111 label=[0x0-0x1a81a8].cc.arduino.Arduino path=/Applications/Arduino.app/Contents/MacOS/JavaApplicationStub flags=1 : LaunchApplicationClient.cp #963 LaunchApplicationWithSpawnViaLaunchD() q=com.apple.root.default-qos
```

Following @richardloxley's steps above (running from outside the applications folder before copying) doesn't seem to change any files or permissions but does indeed solve the problem.