arduino/Arduino #4316: Invalid library found

sasa999, 2015-12-18:
Invalid library found in F:\Arduino\libraries\DFR0299_library: F:\Arduino\libraries\DFR0299_library

How to fix this library

mp3TF.h
```
/*  2014-01-07 by lisper <leyapin@gmail.com>

*  control mp3 module by seirial

*/

#include <Arduino.h>

class mp3TF{ 

  HardwareSerial* mySerial;

public :

void init (HardwareSerial*);

void sendCmd (uint8_t *);

  void play();

void play (uint16_t index);

void play (int index);

  void pause();

  void stop();

  void next ();

  void previous();

void setVolume( byte vol );

};

```

```
mp3TF.cpp

/*  2014-01-07 by lisper <leyapin@gmail.com>

*  control mp3 module by seirial

*/

#include <Arduino.h>

#include "mp3TF.h"

//

void mp3TF::init (HardwareSerial* theSerial) {

mySerial = theSerial;

}

//

void mp3TF::sendCmd (uint8_t *cmd) {

uint8_t checksum = 0;

for (int i=2; i<8; i++) {

checksum += cmd;

}

cmd[8] = (uint8_t)~checksum;

for (int i=0; i<10; i++) { //send cmd

(*mySerial).write (cmd);

}

}

//

void mp3TF: : play() {

static uint8_t play_cmd [10] = { 0X7E, 0xFF, 0x06, 0X0D, 00, 00, 00, 0xFE, 0xee, 0XEF};

sendCmd (play_cmd);

}

//

void mp3TF: : play(uint16_t index) {

static uint8_t play_cmd [10] = { 0X7E, 0xFF, 0x06, 0X03, 00, 00, 00, 0xFE, 0xee, 0XEF};

play_cmd[5] = (uint8_t)(index >> 8);

play_cmd[6] = (uint8_t)(index);

sendCmd (play_cmd);

}

//

void mp3TF: : play(int index) {

play ((uint16_t)index);

}

//

void mp3TF: : pause() {

static uint8_t pause_cmd [] = { 0X7E, 0xFF, 0x06, 0X0E, 00, 00, 00, 0xFE, 0xED, 0XEF};

sendCmd (pause_cmd);

}

//

void mp3TF::stop() {

static uint8_t stop_cmd [] = { 0X7E, 0xFF, 0x16, 0X0E, 00, 00, 00, 0xFE, 0xED, 0XEF};

sendCmd (stop_cmd);

}

//

void mp3TF::next () {

static uint8_t next_cmd [] = { 0X7E, 0xFF, 0x06, 0X01, 00, 00, 00, 0xFE, 0xFA, 0XEF};

sendCmd(next_cmd);

}

//

void mp3TF: : previous() {

static uint8_t previous_cmd [] = { 0X7E, 0xFF, 0x06, 0X02, 00, 00, 00, 0xFE, 0xF9, 0XEF};

sendCmd (previous_cmd);

}

```

per1234, 2015-12-19:
I believe the problem is you have incorrectly installed that library. The library files should be located under:
`F:\Arduino\libraries\mp3TF`
I'm guessing you have them under:
`F:\Arduino\libraries\DFR0299_library\Arduino_library\mp3TF`

sasa999, 2015-12-19:
You are right. Thanks!
SOLVED

closed 2015-12-21 by cmaglie