arduino/Arduino #4320: Delay yield

vbextreme, 2015-12-19:
While writing a port official scheduler I realized that the delay() function has a bug.
this is official delay() function
```
void delay(unsigned long ms)
{
	uint16_t start = (uint16_t)micros();

	while (ms > 0) {
		yield();
		if (((uint16_t)micros() - start) >= 1000) {
			ms--;
			start += 1000;
		}
	}
}
```
This code assumes that the task called from delay lasts less than a millisecond , otherwise you can get up to an infinite loop.
simple solution:
```
void myDelay(unsigned long ms)
{
      if ( 0 == ms ) return;
      
      uint32_t sm = micros();
      uint32_t sM = millis();

      while( millis() - sM < ms -1 ) yield();
      ms *= 1000;
      while( micros() - sm < ms - 1);
}
```
It works with the same precision but should be free of bugs.

cmaglie, 2015-12-21:
good catch,

I prefer the second version (using `uint32` instead of `uint16` and the `while` instead of the `if`) because the sequence:

```C++
      uint32_t sm = micros();
      uint32_t sM = millis();
```

it's still subject to race conditions (millis() may advance when the execution is in the middle of the two instruction). Would you like to make a pull request for this one?

matthijskooijman, 2015-12-21:
I also prefer the second one (I think you edited your post to add it? I don't remember seeing it in the e-mail notification).

> it's still subject to race conditions (millis() may advance when the execution is in the middle of the two instruction). Would you like to make a pull request for this one?

What race condition exactly? You mention `millis()`, but the second version doesn't use `millis()` at all? I can't see any race condition here?

cmaglie, 2015-12-21:
> What race condition exactly? You mention millis(), but the second version doesn't use millis() at all? I can't see any race condition here?

The second version if fine, I'm referring to the first version.

If the millis() update ISR fires exactly in the middle of the two instruction:

```C++
      uint32_t sm = micros();
      uint32_t sM = millis();
```

you can get something like:
sm == 100999
sM == 101

while you really wanted:
sm == 100999
sM == 100

or:
sm == 101000
sM == 101

matthijskooijman, 2015-12-21:
I don't actually think that your example would cause any problem? In the end, the micros() value defines the exact moment of returning. I do think that the second loop should be `< ms` instead of `< ms - 1`, though. In any case, I think the other implementation is better, also because the first one is limited to 2^32 us, while the second can do 2^32 ms.

vbextreme, 2015-12-22:
if you have five task running,which consume two milliseconds can this be
```
start = micros()/*0*/;

while( ms()/*2ms*/ )
{
    yield();/* run another 4 task, ellapse 8ms*/
    if ( micros()/*> 8ms*/ - start/*0*/ > 1/*ms*/ )
    {
        --ms;/*1*/
        start += 1 /*ms*/;
        /* big problem! ellapse 8 ms but remove olny one ms and increment start for only one milliseconds.*/ 
    }
} /*LOOP*/
```

The start losing information causing long waits unexpected.
If the yield is more long, for example approximate 65ms can have a infinite loop because it goes to fall back always near the start.
Both issues are real, a task or set of tasks can last more than a ms.
Since the function delayMicroseconds() does not implement the functionality of the yield() a task can get to have lasting (~ 0L) us, therefore a time of 65ms is acceptable.

The first solution that I posted is simple and subbivide the problem into two subproblems.
We will use Mircos() to try to get a delay() relatively accurately, and millis() just as extra time to address other tasks.
Then we're going to pick the exact start in (us) and a monitoring time (ms).
Now we enter the loop until it has elapsed the time right -1, in this way the subsequent loops will have the time necessary to increase the precision, the -1 in this case assumes a (us) lost in this whole process, in reality need a -4.
No race condition in the millis (), is not critical.

The second code I posted it at a later time, trying to elimiare the bug but using the "old" code delay() so as not to disrupt anything.

Theoretically it is more precise and secure the first version but in this context I think are equivalent and therefore I would opt for the second code fragment.

If confirmed execute the pull request .
These days I have little time , sorry for the delayed response

VogonJ, 2015-12-23:
Iis this delay thing also an issue for Due Scheduler threads ?
Or to which scheduler is this topic related ?

vbextreme, 2015-12-23:
You have a DUE?
Create One task and use micros for get real time delay(2), add second task and use delayMicroseconds(65536).
After exit on delay in first task send on serial real time of delay.

we look what happens.

vbextreme, 2015-12-24:
try this code
```
#include <Scheduler.h>

void loop2() {
  uint32_t ts,te;

  Serial.print("delay(10) == ");
  digitalWrite(13, HIGH);
  
  ts = micros();
  delay(1000);
  te = micros();
  
  digitalWrite(13, LOW);
  Serial.print(te-ts);
  Serial.print("us == ");
  Serial.print((te-ts)/1000);
  Serial.println("ms");
  delayMicroseconds(800);
}

void loop1() {
  
  digitalWrite(13, LOW);
  delayMicroseconds(65535);
  digitalWrite(13, HIGH);
  yield();
}

void setup() {
  Serial.begin(9600);
  pinMode(13, OUTPUT);

  Scheduler.startLoop(loop1);
  Scheduler.startLoop(loop2);
}

void loop() {
  yield();
}
```

VogonJ, 2015-12-26:
hello,
I don't understand what actually is expected to happen - 
I ran the code  (by Serial.bein(115200) as usual) and it shows

delay(1000) == 1049914us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049929us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049929us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049929us == 1049ms
delay(1000) == 1049926us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049928us == 1049ms
delay(1000) == 1049928us == 1049ms

matthijskooijman, 2015-12-26:
@VogonJ, what sketch did you use to generate this output? Perhaps the generation of the debug output takes 49ms there? Did you test with this PR applied, or also without it? Did this PR change your debug output?

VogonJ, 2015-12-26:
it's the code that was posted posted above, just Serial.begin(115200) instead of 9600.

PaulStoffregen, 2015-12-26:
This is the nature of cooperative scheduling.  Your loop1() function calls delayMicroseconds(65535), which doesn't yield.  Other tasks will be blocked during that delay.

VogonJ, 2015-12-26:
so what is the non-blocking way to use delay()  for the Due Scheduler?
(Finally some tasks have to be delayed indispensably for certain purposes)

NicoHood, 2015-12-26:
Since when does the avr core support the port scheduler?
This is not documented. I am confused??
https://www.arduino.cc/en/Reference/Scheduler

Testato, 2015-12-26:
This is only a PR for a bug that VBExtreme discovered during his development of the porting for the AVR of the SAM Scheduler.
On his website he released the first version of the porting (i tested it and working), at the end of the work i think that VBExtreme will create a new PR for the Scheluder Library.
This will be a very interesting news on Arduino AVR Core.

If you want test the porting:
http://vbextreme.netai.net/2015/12/26/embedded-arduino-scheduler-patre-1-1/

VogonJ, 2015-12-26:
sorry, my English is VERY poor...

DIUC, that the delay() issue is fixed for the AVR core,

and there is the same delay() issue for the ARM core,

but it's not fixed for the ARM core (Scheduler) yet, but work about the Scheduler is in progress ?

vbextreme, 2015-12-26:
@NicoHood , @Testato , not introduce speeches are missing on issues. Talk about Arduino Due, Official Scheduler and Delay/yield.

@PaulStoffregen , on the test of @VogonJ time return 1049ms and NOT 1065ms, for this reason I think that loop1 not being called in loop2.

try this and see to take away some doubts.
```
#include <Scheduler.h>

#define dbg(S) do{ Serial.print(__FUNCTION__); Serial.print("() "); Serial.println(S);}while(0)

void loop1() {
  
  dbg("in");
  
  uint32_t ts,te;

  Serial.print("delay(10) == ");
  digitalWrite(13, HIGH);
  
  ts = micros();
  delay(1000);
  te = micros();
  
  digitalWrite(13, LOW);
  Serial.print(te-ts);
  Serial.print("us == ");
  Serial.print((te-ts)/1000);
  Serial.println("ms");
  delayMicroseconds(800);
  dbg("out");
}

void loop() {
  dbg("in");
  digitalWrite(13, LOW);

  uint32_t st = micros();
  while( micros() - st < 6525);
  
  digitalWrite(13, HIGH);
  yield();
  dbg("out");
}

void setup() {
  Serial.begin(9600);
  pinMode(13, OUTPUT);

  Scheduler.startLoop(loop1);
}
```

vbextreme, 2015-12-27:
sorry, is not bug for Arduino DUE.
This is a delay for SAM
```
void delay( uint32_t ms )
{
    if (ms == 0)
        return;
    uint32_t start = GetTickCount();
    do {
        yield();
    } while (GetTickCount() - start < ms);
}
```

is bug anly for avr.

VogonJ, 2015-12-27:
please, just for my understanding:
is delay() for the DUE inside of loop() or inside of Scheduler tasks (loop1(), loop2(),...) now a blocking delay or a non-blocking delay ?

vbextreme, 2015-12-27:
I made a mistake , I thought that the delay() function was the same for everyone , but for Arduino DUE is different .
I reported the bug only occurs if you are using yield() on Arduino Uno,aka avr328p, and yield() takes more one milliseconds.

VogonJ, 2015-12-27:
so is delay() for the DUE inside of loop() or inside of Scheduler tasks (loop1(), loop2(),...) now a blocking delay or a non-blocking delay ?

matthijskooijman, 2015-12-27:
> so is delay() for the DUE inside of loop() or inside of Scheduler tasks (loop1(), loop2(),...) now a blocking delay or a non-blocking delay ?

I'm not sure, and I think @vbextreme doesn't know either (since he's working with AVR). AFAICS, `delay()` is always blocking, but due to the `yield()` it can run other scheduler tasks during the delay. I'm not sure if calling `yield()` or `delay()` inside a scheduler task is allowed, if the scheduler schedules the same task again, things could break (but perhaps the scheduler takes care of this already, I don't know). In any case, your question is really off-topic from the issue at hand, which is the `delay()` implementation on AVR. So please refrain from further discussion about this here, questions like these are better placed on the Arduino forums.

vbextreme, 2015-12-27:
@matthijskooijman, not only DUE can call yield, you can find this function in arduino UNO core and you can replace as does the DUE.
example to view a bug:
```
void yield(void)
{
    delayMicroseconds(65535);
}

void setup() {
  Serial.begin(9600);
}

void loop() {
  
  uint32_t ts,te;

  Serial.print("delay(1000) == ");
  ts = micros();
  delay(1000);
  te = micros();
  
  Serial.print(te-ts);
  Serial.print("us == ");
  Serial.print((te-ts)/1000);
  Serial.println("ms");
}
```
and output is:
```
delay(1000) == 16800056us == 16800ms
delay(1000) == 16783744us == 16783ms
delay(1000) == 16783752us == 16783ms
delay(1000) == 16783752us == 16783ms
delay(1000) == 16783748us == 16783ms
delay(1000) == 16783748us == 16783ms
delay(1000) == 16783752us == 16783ms
```

you can see that the yield() is invoked automatically by delay() , just as it does the DUE.
If you want implement a scheduler for arduino UNO like Ufficial Scheduler you can't! because on arduino UNO have a bug!
Now what is the yield() in UNO if not then we can use it?
or takes off or removing the bug.

vbextreme, 2015-12-27:
so, if you edit wiring.c->delay() with myDelay the output is:
```
delay(1000) == 1010412us == 1010ms
delay(1000) == 1010404us == 1010ms
delay(1000) == 1010412us == 1010ms
delay(1000) == 1010404us == 1010ms
delay(1000) == 1010412us == 1010ms
delay(1000) == 1010412us == 1010ms
```

matthijskooijman, 2015-12-27:
@vbextreme, Yes, that's exactly how I understood it already.

VogonJ, 2015-12-27:
Too many ambiguous words, so again please:

I know for sure that even the Due Scheduler examples are using delay(), but that didn't resolve if it's blocking or not.
So please in 1 (ONE) word:

blocking for the Due Scheduler or not - for sure (!) ?
YES ?
NO ?

vbextreme, 2015-12-27:
@VogonJ  
on Arduino 2:
if you use more tasks and in the task call delay(), NOT block, because current task call another task in waiting state.
on Arduino 1:
if ( you not edit a scheduler.cpp ) you can't use a multi task and delay() block.
else if ( you edit a scheduler.cpp ) you can run multi task but 
if ( you not edit a delay() ) your sketch fails.
else if ( you edit a delay() ) arduino 1 works as the 2

@matthijskooijman You think would appreciate a pull request for porting the scheduler?

VogonJ, 2015-12-27:
not blocking on DUE, thank you, 

So  PaulStoffregen's statement is obviously false - that was which has been confusing me:
"This is the nature of cooperative scheduling. Your loop1() function calls delayMicroseconds(65535), which doesn't yield. Other tasks will be blocked during that delay." https://github.com/arduino/Arduino/issues/4320#issuecomment-167327339

Testato, 2015-12-27:
+1 for porting the scheduler on AVR core,
Update the official scheduler library, so we have only one lib, is the best way for me

Testato, 2015-12-27:
@VogonJ 
delayMicroseconds is different from delay2

Please ask this question on forum not here

vbextreme, 2015-12-27:
@VogonJ is not false, delay() not block and call another task, but in that example the second task call delayMicroseconds() and this function blocks the parent who called.
ask on forum.

VogonJ, 2015-12-27:
thank you for clarification that delay() is different from delayMicroseconds() !
That was the missing link so far!

PaulStoffregen, 2015-12-27:
> thank you for clarification that delay() is different from delayMicroseconds()

After this clarification, do you still believe "So @PaulStoffregen 's statement is obviously false"?

duff2013, 2015-12-28:
Wow I wonder if @VogonJ can be blocked from these discussion? He's at best annoying like a little dog nipping your ankles and at worst a pug troll that just wants to muddle the conversation for his own amusement. 

Dude I went to your github page and you don't even bother cloning anything just comments, get a life!

peabo7, 2015-12-28:
I disagree.  Everyone here was once non-expert in some project or other (not necessarily Arduino).

What I find frustrating about @VogonJ is his persistent claim that he is incapable of growing into the next level and his demand that someone else figure out what he wants to know.

@VogonJ you could answer the question you asked about delay() and the scheduling library yourself by examining the source code, performing experiments, and arriving at an explanation.  If you had, you would be much more accomplished at using the Arduino platform than you are now.

PaulStoffregen, 2015-12-28:
Like others who contribute to Arduino, I subscribe to Arduino's issue tracker.  Every message generates an email message.  Recently, a huge number of these messages have little or nothing to do with improving Arduino.  @VogonJ is one of the worst offenders, essentially treating Arduino's issue tracker as a tech support forum, with an expectation that Arduino's developers will devote their engineering time which should be spent improving Arduino to answer his specific tech questions.

Usually I just unsubscribe from each individual issue when he begins posting.  Issues like this one are a difficult case, where most of the messages are off-topic, but the issue is an area where I'm concerned about Arduino's performance.

I really wish users like @VogonJ who consistently abuse Arduino's issue tracker could be banned.  They waste everyone's valuable time with their selfish requests which belong on Arduino's forum, not the issue tracker.

VogonJ, 2015-12-28:
my question were simple and it had been easy to give clear and short answers:
Is this delay thing also an issue for Due Scheduler threads ?
Or to which scheduler is this topic related ?
Is delay blocking for the Due or not.

Now read closely the answeres which had been given: ambiguous, indeterminate, vague, and unspecific  
- finally I asked just about the Due and just about delay() and not about delayMicroseconds.

So if anyone is not able to answer clearly he must not complain about follow-up questions.

matthijskooijman, 2015-12-28:
@VogonJ, the main point is that this issue tracker is the wrong place for your questions in the first place. As for the vagueness of the answers, your question is similarly vague. You ask about "blocking", but what is blocking according to you? None of the delay functions return before the delay is finished, so in that sense they are all blocking. What you meant was if other scheduled tasks can run during the delay, which is the case for `delay()`, not `delayMicroseconds()`.

Anyway, the take-away point here is that this issue tracker is not meant for questions, but for bug reports and feature suggestions.

VogonJ, 2015-12-28:
I have issues about Due delay, and so my question
"Is this delay thing also an issue for Due Scheduler threads ? (NO.)"
was definitely  on-topic.

matthijskooijman, 2015-12-28:
@VogonJ, Ah right, your initial comments were indeed on-topic, now I re-read them with some more context. Later questions were not really, though I can see how they arose from the earlier discussion, so I stand corrected. In general though, you have a history for asking questions on the issue tracker that do not really belong here, so my point still stands.

VogonJ, 2015-12-28:
Apologies are accepted and I will reflect more insively about future questions of understanding.

Anayway, I am still curious if duff2013 perhaps could be banned because of hostile, insulting, attacking and offensive trolling.

PaulStoffregen, 2015-12-28:
My inbox Arduino folder currently has 361 messages from "VogonJ" and "VogonJeltz", even though I've unsubscribed from most issues where he's posted.  This is a consistent pattern of off-topic, time-wasting messages.

VogonJ, 2015-12-28:
I have read lots of complaints about your rude attitude, and I don't know what's more annoying.

Anyway, I am curious how to reimburse a prosecutorial criminal complaint for insulting against duff2013.

matthijskooijman, 2015-12-28:
@duff2013, even though I sympathize with the message in your comment, your wording is indeed offensive and IMHO inappropriate. Please keep that in mind when writing future comments.

With that said, I'm locking this issue, since I do not believe further responses will be productive. For discussion about the issue and fix at hand, feel free to comment on the pullrequest instead.

matthijskooijman, 2015-12-28:
This issue is fixed by #4349.

closed 2015-12-28 by matthijskooijman