arduino/Arduino #4321: Please add: SwissHandmade_Arduino-Examples

chatelao, 2015-12-19:
Please add my library:
- https://github.com/chatelao/SwissHandmade_Arduino-Examples

It will include more and more samples with more configurations possibilities than the original arduino samples.

NicoHood, 2015-12-19:
Are you going to extend the "library". Not sure if this is really needed at this stage.

chatelao, 2015-12-19:
I will add more step by step. Next are some changes on the LCD samples. If
you think it is better to keep it private for a while it is okay too. I was
first of all tired to modify the Blink for every platform without D13 (E.g.
attinies with only 4 gpios) . A define of the LED_PIN would fix most of the
problem in the IDE.

closed 2015-12-20 by chatelao