arduino/Arduino #4322: Add MiniPirate library, a command line for Arduino

chatelao, 2015-12-20:
Pleas add the "MiniPirate" library. It contains a serial command line tool for Arduino running on any OS with a serial monitor:

- https://github.com/chatelao/MiniPirate

For more informations see these articles of the older, non-library edition:
- Hackaday: https://github.com/chatelao/MiniPirate
- Instructable: http://www.instructables.com/id/Arduino-comand-line-tool-MiniPirate/

cmaglie, 2015-12-21:
Done, thanks!

closed 2015-12-21 by cmaglie

chatelao, 2015-12-21:
Thanks, cannot wait next Sunday night

cmaglie, 2015-12-21:
You don't need to wait, it's already there :-)
(the Sunday job is for clean-up, i.e. removing old versions or fixing errors on already published libraries)

chatelao, 2015-12-21:
Help: I tried to search it after restarting the IDE, but it didn't show up. Any idea how I could debug the process?

cmaglie, 2015-12-21:
Just writing the first letters "Minip".. in the search box should do:

[image]

Maybe your IDE is not able to update the `library_index.json` file for some reason?

chatelao, 2015-12-21:
Thanks for your nice answer, I will try it again. Now I know the name of the file to search for :-)