arduino/Arduino #4324: LiquidCrystal.print() not documented accurately

rawtaz, 2015-12-21:
The second argument to LiquidCrystal.print() can be used to specify the number of decimals that should be printed when giving a float as the first argument. But this is not described or mentioned in https://www.arduino.cc/en/Reference/LiquidCrystalPrint, it probably should be.

per1234, 2015-12-22:
@agdl maybe you could just create reference pages for the Print functions and then link to those pages from the relevant reference pages of each library that uses the Print class. The library specific reference pages would contain a description of what the function does in that library and the link to the Print function reference page provides usage details.

q2dg, 2015-12-26:
I'd extent @perl1234 idea to Stream functions too.

per1234, 2017-07-06:
Even though this one came first, I'm closing it as a duplicate of https://github.com/arduino/Arduino/issues/5588 because that issue has slightly more information. I am also transferring the assignment of agdl to #5588, which is currently unassigned.

closed 2017-07-06 by per1234