arduino/Arduino #4330: IDE / OSX / ZERO remove the openocd.bin popup

extrapixel, 2015-12-22:
[image]
From a user perspective it's not obvious what this binary is and why it needs to accept incoming connections. Plus, it's annoying to click it away with every upload to the ZERO board.

matthijskooijman, 2015-12-22:
I think openocd opens up a port on localhost for three different protocols (gdbserver, telnet and one more). One of them is disabled in the openocd script, but not the others. They should all be disabled, but possibly not in the config file but on the commandline. I'll pick this one up along with my other openocd improvements.

sandeepmistry, 2015-12-22:
@matthijskooijman I started looking at this briefly.

Although the OpenOCD documentation suggests you can disable the ```gdb``` port via ```gdb_port disabled``` is doesn't look like this option actually works. An error is reported ```Error: could not open disabled```.

I can look into submitting a patch upstream for this, unless you want to?

matthijskooijman, 2015-12-22:
Hm, I seem to vaguely remember some discrepancies between docs and code in this area. Perhaps gdp_port 0 works? It would be worth submitting a bug report or patch upstream if actual behaviour does not match docs, though. Feel free to do so, then I'll think about how / where to put the actual disabling directive :-)

sandeepmistry, 2015-12-22:
Unfortunately ```gdp_port 0``` appears to bind to a random port.

Sounds good, I'll keep you posted on the patch progress here.

matthijskooijman, 2015-12-22:
You might want to check the other options as well, IIRC openocd binds to three ports by default (telnet_port, gdb_port and one other).

sandeepmistry, 2015-12-22:
It's definitely the ```gdb_port``` the other port is ```tcl_port```.

This is the commit in OpenOCD which removed the 0 / disabled option for ```gdb_port```: http://sourceforge.net/p/openocd/code/ci/6c137a2fc0bf53b9c0b8eda51e6f5361552b0112/

matthijskooijman, 2015-12-22:
Not sure if I understand that commit fully, but it appears that tcl_port and telnet_port got special handling for "disabled", but gdb_port did not. In fact, I suspect that previously, "disabled" worked because the integer parsing failed and returned 0, which was interpreted as disabled for all options?

sandeepmistry, 2015-12-23:
I've submitted the following OpenOCD patch upstream: http://openocd.zylin.com/#/c/3175/

We can add the parameter in the SAMD core's ```platform.txt``` or ```arduino_zero.cfg```, when we have a new build that supports it.

sandeepmistry, 2016-10-17:
@cmaglie http://openocd.zylin.com/#/c/3175/ has been merged so we can pull in the change into the Arduino version of openocd.