arduino/Arduino #4332: One wire library still in v2.3.0 - causes useless warning

vicnevicne, 2015-12-22:
The version of the OneWire library proposed by the library manager is still v2.3.0, which causes the IDE to raise the following warning on compile :

WARNING: Category '' in library OneWire is not valid. Setting to 'Uncategorized'

This fix (and others) is included in OneWire lib v2.3.1 which was released on Dec. 6th - https://github.com/PaulStoffregen/OneWire/releases - so the IDE should link to v2.3.1 now.

matthijskooijman, 2015-12-22:
I had a look at that repository, it looks like it has a proper tag, which also contains the proper version number in its library.properties file, so it should be picked up. I can't look into the server side system for errors, though.

vicnevicne, 2015-12-22:
Hi, Matthijs, and thanks for looking into this.
Do you know how the "standard" libraries are referenced in the IDE ? Is there a list of urls in a config file somewhere or is the list fetched online ? Could it be a caching issue ? 
Best regards. Vicne

cmaglie, 2015-12-23:
@vicnevicne 
it seems that the library has been updated in the library manager:

[image]

maybe your PC is unable to update the `library_index.json` file?
(if you're on windows you can find it in `%LOCALAPPDATA%/Arduino15` folder)

vicnevicne, 2015-12-23:
Hi, and thanks for helping. 
That's weird : I can find the file you mention and gets updated by the IDE, but the contents still references only 2.3.0.
Here are the steps I just did :
1) Exit IDE
2) Delete the old library version. I admit I found no procedure about how to uninsall a lib so I just delete both :
"C:\Users\<me>\AppData\Local\Arduino15\staging\libraries\OneWire-2.3.0-github.zip"
and :
"C:\Users\<me>\Documents\Arduino\libraries\OneWire2.3.0\"
3) Restart IDE
4) Sketch > Include library > Manage libraries > Filter on "OneW" > Click on OneWire 
(I'm just getting an "Install" button with no choice of version)
5) Click "Install"
=> Shows version 3.2.0 :

[image]

The library_index.json file has been updated (see timestamp) but only references v2.3.0 :

[image]

What is the URL the library_index.json gets downloaded from ? Could it be that a config file somewhere on my machine still points to an outdated URL or something ? Or could it be that the URL is somehow unreachable which would cause the old version to be reused ?
Best Regards. Vicne

vicnevicne, 2015-12-24:
OK. Did more tests and it seems clear now that this is due to issue #4272 
The gzip file did not get updated (still dated August, see above).
After deleting it, it was re-downloaded on next use of the library manager and I was given the option to update OneWire to 2.3.1.
So I guess this issue can be marked Duplicate, but that puts more pressure towards #4272 :-)
BR. Vicne

cmaglie, 2015-12-30:
Fixed by #4361

closed 2015-12-30 by cmaglie

PaulStoffregen, 2016-01-06:
I recently released OneWire 2.3.2, with support for Arduino Zero & Arduino 101 (Intel Curie).

Can anyone confirm if they're seeing the latest 2.3.2 release in the library manager?

vicnevicne, 2016-01-06:
Confirming. 
I got a popup yesterday night informing me that an update of one of an installed library was available. Opened the library manager and it I could select version 2.3.2 for OneWire and perform the update.
So all good for me :-)