arduino/Arduino #4333: file ".c" extension

Testato, 2015-12-22:
Only ".h+.cpp" extension used under a sketch folder work.

If you try to use a .c extension for a file the Arduino IDE print some error,

Try it whit this example, it do not work but if you change the test.c to test.cpp it will work

https://drive.google.com/open?id=0Bw27MYXTb2FXR0ZpZDVLSDh6cDA

NicoHood, 2015-12-22:
Ha, i had the same issue today. Not sure if the issue was caused by us or the compiler.
You need to add extern "C" definitions around the included header files. Then it works.

However I noticed another error: If you rename the files the other is kept in the temporary directly for compiling and sometimes leftover. Saving fixes this, editing again breaks it again. I have no test for this though.

matthijskooijman, 2015-12-23:
@Testato, as Nicohood indicated, extern "C" is needed when mixing C and C++ (.ino files are compiled as C++). Google keywords are name mangling and calling conventions. So, this behaves as expected and I will close this ticket. If we misunderstood and the problem has another cause, feel free to leave a comment so we can reopen it. If you do, be sure to include the actual error message as well.

@NicoHood, please open a new issue for that? I think I know what's going on and I recall seeing code to fix that for deleting files, but not for renaming apparently. I've been digging into that part of the code, so I can probably pick that one up.

closed 2015-12-23 by matthijskooijman

lmihalkovic, 2015-12-23:
@NicoHood i can confirm the behavior. It is the first issue i ran into when i started using the ide a couple months ago.