arduino/Arduino #4334: preprocessor allows errors to be compiled

i-make-robots, 2015-12-22:
#if false
#include <this_file_does_not_exist.h>
#endif

void setup() {}
void loop() {}

should not create a linker error... but it does.

per1234, 2015-12-23:
Using which version of the Arduino IDE? It compiles fine for me with 1.6.7 and the hourly build.

vbextreme, 2015-12-23:
1.6.6 work

matthijskooijman, 2015-12-23:
There have been some improvements in recent versions, so be sure to test the hourly build. Since @per1234 tested this, I'm closing this issue as already fixed in the latest version. @i-make-robots, if you test the latest hourly build and it is still broken for you, leave a comment here and we can reopen. Also, be sure to include the actual error message when reporting issues, that makes it easier to figure out a problem.

closed 2015-12-23 by matthijskooijman

i-make-robots, 2015-12-23:
Sorry, my bad. My first invalid ticket :)