arduino/Arduino #4335: Renaming C to CPP gives errors

NicoHood, 2015-12-23:
@matthijskooijman @testato @lmihalkovic 
https://github.com/arduino/Arduino/issues/4333#issuecomment-166728159

If you rename the c files to cpp (or vice versa) the other is kept in the temporary directly for compiling and sometimes leftover. Saving fixes this, editing again breaks it again. I have no test for this though.

matthijskooijman, 2015-12-23:
When you have unsaved changes, the IDE writes a copy of the sketch to a temporary dir, and lets arduino-builder compile that. When there are no unsaved changes, arduino-builder is called on the original sketch dir instead. Since the rename is not done on the temporary copy, it breaks, and saving it fixes it because then the original sketch dir is used again. I suspect that simply deleting the temporary copy after each compilation would be best here, since every new compilation completely copies it again anyway.

Testato, 2015-12-23:
> #4333 (comment)

This happen  when you rename the file by OS and the file is already opened
in the IDE. If you rename and reboot the ide the problem disappear.

It is a bug or a normal behaviour ?
The IDE must refresh and re-read the filename+extension ?

One of the solution can be enable renaming inside IDE

matthijskooijman, 2015-12-23:
I actually think the  bug NicoHood is describing happens when you rename the file withing the IDE itself. If you rename the file externally, then indeed, the IDE will not know about this and you must restart the IDE (or re-open the sketch) to make it find out. In general, you should not modify your sketch externally, unless "external editor" is set. At some point, it might make sense to let the IDE automatically check for external modifications, even when external editor is disabled, but this is not the subject of this issue :-)

Testato, 2015-12-23:
So is it posdible change a file name inside ide ? From pippo.c to pippo.cpp or pippo2.c ?

lmihalkovic, 2015-12-23:
IMHO this is a real annoyance that makes the ide feel like it comes straight from the 80's... which can be fixed with only a handfull of loc.

matthijskooijman, 2015-12-23:
@Testato, Yes, there is a "rename tab" option in the dropdown to the right of the tabs (little arrow/triangle pointing down).

NicoHood, 2015-12-23:
I used the rename tab in this case.
Not sure if deleting all then disables the "do not recompile already compiled files" optimization.

matthijskooijman, 2015-12-23:
Hm. I think it will not disable that optimization, but I'm also not really sure why that optimization works at all when there's unsaved changes. In any case, in the process there's (at most) four copies of the sketch files:
 1. The original files in your sketchbook
 2. The files loaded into memory in the editor. If there's unsaved changes, these are different from 2.
 3. A temporary save of the sketch. This is generated by the IDE when there are unsaved changes, to allow passing the in-memory modifications to arduino-builder.
 4. A copy in the final build folder. The .ino files are modified here with prototypes, other files are copied as-is.

When calling arduino-builder, 1. is passed if possible, but if there are modifications, 3. is generated and passed. 4. is always generated by arduino-builder.

If there are no modifications, arduino-builder can compare timestamps between 1. and the .o files in .4 and reuse them if unchanged (directory 4 is kept around until the IDE is closed and deleted then).

However, when 3. is generated, those will always get fresh timestamps now. I thought that perhaps arduino-builder was passed both the original sketch as well as 3., but it seems it just gets 3. So I'm not so sure why how it can detect that a file was not changed, but somehow it does (even when I delete directory 3. manually). I'm a bit puzzled about this.

In any case, it seems like an improvement over the current mechanism would be to only write the modified files to 3. and then pass both 1. and 3. to arduino-builder. It will then use files from 3. when they exist and 1. otherwise. This allows arduino-builder to see the original timestamps on unmodified files. On modified files, this will always see an updated timestamp (so these will always be recompiled, even if they were not changed since the last compile), but that seems ok. A big advantage of this approach is that any unmodified files, or additional files (in the sketch directory, but not loaded in the editor) do not need to copied into 3., saving one copy (they will still need to be copied into 4.)

matthijskooijman, 2015-12-23:
On additional reason for the last change proposed above, is that now if you compile without modified files and then with modified files, the sketch path changes (as far as arduino-builder can see) so arduino-builder recompiles *everything*.

matthijskooijman, 2015-12-23:
Ah, figured out why arduino-builder does not currently recompiles all sketch files when some files are modified. Arduino builder currently copies files from the sketch directory (1. or 3. above) into the build directory (4. above), but first compares their content. If the content has not changed, the files are not copied into the build directory. Then the actual compilation looks at the timestamps of the *source* files in the *build* directory, which will not have changed if the contents of a file did not change. So, deleting directory 3. in the IDE after every compilation should not affect this optimization currently. I still think we should consider my above suggestion, but for now we could start with fixing this on the IDE side (deleting the entire directory 3. should also happen when it just contains changed files).

lmihalkovic, 2015-12-23:
Dunno what ou will do for your version, but i am rewriting this. I want ro keep track of files as they get modified and compile only what nedsto be compiled. What i saw is that everybit of knowledge that ever exists at layer in the chain gets wiped when going down. my situation is made worse by the fact that i support local libraries inside a sketch. I am also leaning towards caches per architecture (nobody has to know that ido it) so that a sketch can tested accross boards without encurring compile-the-world each time. In the end this is not impacting builder as i have my own compiler-driver (i can switch between builder and my own).

matthijskooijman, 2015-12-23:
Any suggestions on how to remove the need for directory 3? Passing file contents through stdin as JSON or something like that? In the end, the modified contents will have to end up in a file for gcc to work with it (or you can probably let gcc compile from stdin, but I'm not sure if that's the best way to handle this...).

Testato, 2015-12-23:
Sorry for the noise, but a question:
In the 1.6.7 official it is not supported use a library inside thev sketch folder ?
Actually if i update a lib by library manager i loose my lib modification.

cmaglie, 2015-12-23:
> Actually if i update a lib by library manager i loose my lib modification.

If you're developing a library you probably don't want to use the lib manager at all, but clone it with git instead (that has a much better tracking capabilities so your changes can be easily maintained in sync with upstream development)

BTW this question is **totally** offtopic here, please open another issue or comment on an already existing one.

matthijskooijman, 2015-12-28:
@NicoHood, I just created a PR to fix this. Could you perhaps try it to confirm it fixes your problem?

PR #4358 (2016-01-19): Fix compile errors after renaming files, related to unsaved changes in the sketch
Fix #4335 by always deleting the temporary sketch copy after every build. With that, the name of that directory can be random again, instead of needing to be deterministic using a hash of the main sketch filename.
commit 1029e0b7 (2016-01-19): Delete temporary sketch copy after build

When a sketch has unsaved changes, a temporary copy of the sketch is
made with those changes applied. This copy is then passed to
arduino-builder.

Previously, this temporary copy was kept around and only deleted when
the IDE was closed. However, all files were written to it again on every
build, so keeping the old files around did not serve any real purpose.

When a file was renamed in the IDE, the original name would still be
present in the temporary copy, and could cause linker errors because
both were compiled.

This commit makes sure the temporary [... cut]
closed 2016-01-19 by cmaglie