arduino/Arduino #4336: Use precompiled .a library

NicoHood, 2015-12-23:
I try to add a precompiled .a library to my sketch. However it does not use the .a file laying inside a libraries folder.

This is an important feature, since you can compile optimized functions with a makefile and include those later. In my case I wanted to use this library:
http://munacl.cryptojedi.org/atmega.shtml

It could later also make sense to add a more complex USB-Core like LUFA.
.o files would also be nice.

I think we already have .S support, hex export and .a linkage, so why not also add precompiled .a files as feature? I guess I somehow need to add a platform.txt file or so, if anyone can help, please do so. Thanks!

matthijskooijman, 2015-12-23:
See also https://groups.google.com/a/arduino.cc/d/topic/developers/Le0DXv0tm9Y/discussion

matthijskooijman, 2015-12-23:
And #199, which is the imported version of the google code issue, closed by @ffisore with:

> Motivation for linking against binaries is that libraries may not be open source. Arduino core license is GPL, so anything that statically links to it must be (at least) GPL as well

I wouldn't consider that sufficient reason to not add this, though - in the end it's up to whoever distributes things to comply with licenses, and there are certainly situations where including precompiled files does *not* violate GPL.

lmihalkovic, 2015-12-24:
```
Motivation for linking against binaries is that libraries may not be open source. Arduino core license is GPL, so anything that statically links to it must be (at least) GPL as well
```
In this formulation, this is not a true statement. Anyone can privately use GPLed code, compile it with non-GPL code, and not have the roof of their office fall on their head (i.e. I do not have to hang my code out on my balcony to have the right to use it with GPLed code). What cannot be done *distribute a derivative work* of a GPLed piece of work which does not itself comply with the GPL terms.

lmihalkovic, 2015-12-24:
```
I think we already have .S support
```
I am not sure I would call what the IDE does *support*. The files can be compiled, but syntax coloring is random.

jhannemann, 2016-01-06:
While for C, with its simple and quasi-standard ABI, this may not be a problem (GPL aside, that is), C++ name mangling and language evolution means that the ABI may change, even between different versions of the same compiler on the same platform. It actually just recently has in [GCC 5 for C++11](http://developerblog.redhat.com/2015/02/05/gcc5-and-the-c11-abi/).

Long story short, binary libraries are more complex than you think and you should avoid them in open source projects, especially when using C++.

NicoHood, 2016-01-06:
Okay that could make sense. However I am speaking of a C library here. It makes sense to include it because there is no other simple way to compile it with the ide itself. Also I think the c limitation (no c++) is not a problem, since we only want to use this for optimized code which is normally never c++.

Thx for pointing this out though.

per1234, 2017-11-03:
Closed by https://github.com/arduino/arduino-builder/pull/219

The Arduino IDE now supports .a and .so files in libraries. See instructions in the `precompiled` property documentation here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

closed 2017-11-03 by per1234

NickChungVietNam, 2018-08-22:
no

David-Niklas, 2018-12-04:
I got a undefined reference error while using a static library. 
https://forum.arduino.cc/index.php?topic=583299.0
Do you know what i did wrong?
It would be great if you could help me.  Thanks :)

David-Niklas, 2018-12-05:
Thanks a lot!
I did try that but now i get the following error:
exec: "/bin/arm-none-eabi-g++": file does not exist
Error compiling for board Arduino M0 Pro (Programming Port).

I think I do not have the driver for the samd. Therefore I did execute \hardware\samd_beta\1.6.25\drivers/dpinst-amd64.exe
but that didn't work. 

Do you know how I can solve that problem?

facchinm, 2018-12-05:
It means that the beta installation didn't succeed... Which IDE version are you using? Make sure `c:\Users\YourUsername\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4` folder exists, otherwise uninstall and reinstall samd beta folder (anc rename the file of course :slightly_smiling_face: )

David-Niklas, 2018-12-05:
I reinstalled the samd beta folder and know the error does't accrue anymore.
Thanks :)

But know i do have the "undefined reference" error again.
I'm using the IDE version 1.8.7.

facchinm, 2018-12-05:
Make sure you are selecting the right board from the menu (they will be doubled, so enabling verbose compilation and checking the paths is a must)

David-Niklas, 2018-12-05:
I did that and got that as output
[https://github.com/David-Niklas/Lib_error/blob/master/error](url)

facchinm, 2018-12-05:
Ok, now everything makes sense :slightly_smiling_face: 

The builder is adding the right folder `C:\\Users\\LED2BE\\Documents\\Arduino\\libraries\\libaccess\\src\\cortex-m0plus` but it's failing to get the library. 

Automatic detection only works if the library is called `lib` + `libraryname.a` , so in your case the archive would be `liblibaccess.a` (ugly :smile: ).

To work around this, add `-laccess` to `ldflasg` field in `library.properties` and the builder should pick up the right archive.

David-Niklas, 2018-12-05:
Awesome!! Thanks a lot.

If i do have multiple libraries: libaccess2.a libaccess2.a libaccess3.a libaccess4.a
Do I need to write the ldflags in the library.propertie file like that?

ldflags=-laccess1.a 
ldflags=-laccess2.a
ldflags=-laccess3.a
ldflags=-laccess4.a

facchinm, 2018-12-06:
A single flag with `ldflags=-laccess1 -laccess2 -laccess3 ... ` (comma separated) will work :slightly_smiling_face:

David-Niklas, 2018-12-06:
Thanks for you help :)
You really did help me a lot!!