arduino/Arduino #4337: Matched brace disappears in some cases

ut2uz, 2015-12-23:
Try to put the cursor before last brace in the following simple example and move cursor forward:

void loop() 
{
}

The upper brace disappears.

If one of matched braces is not in the same column, then OK:
void loop() 
 {
}

The same problem exists for any kind of matched brackets/braces

sandeepmistry, 2015-12-23:
@ut2uz which version of the IDE and OS are you using?

With 1.6.7 on OS X 10.11 I get a white box on the first bracket. Is this what you are seeing?

[image]
[image]

ut2uz, 2015-12-23:
@sandeepmistry , I think this problem exists since 1.6.5, the last is arduino-nightly 1.6.8. The OS is W10. Loop() looks like at your second picture.

per1234, 2015-12-24:
It doesn't only happen when the braces are in the same column. Arduino IDE 1.6.8 2015/12/21 03:51 with Windows 7 64 bit:
[image]
[image]
[image]

VaclavSal, 2016-01-06:
I would be interested looking into this at the source code. 
Since this would be my first attempt I have no clue how / where to find the source code.
Any help in  just locating the source code would be sufficient for now.

matthijskooijman, 2016-01-06:
@VaclavSal, I think that the matching of braces is taken  care of by the RSyntaxTextArea library / component we use. It is built here: https://github.com/arduino/Arduino/blob/master/app/src/processing/app/Editor.java#L1034-L1070 and some related bits of code are here: https://github.com/arduino/Arduino/tree/master/app/src/processing/app/syntax

The rsyntaxtextarea source code itself can be found at https://github.com/bobbylight/RSyntaxTextArea though in Arduino we currently use a slightly modified version (source is here: https://github.com/arduino/RSyntaxTextArea). In the Arduino source tree, RSTA is included as a .jar file (compiled java code) here: https://github.com/arduino/Arduino/blob/master/app/lib/rsyntaxtextarea-2.5.8.1%2Barduino.jar

I hope this helps you find your way around, though it might be a bit tricky as your first introduction, because the code is split between two projects somewhat.

VaclavSal, 2016-01-06:
Thanks, I'll give it a try.

nickgammon, 2016-01-17:
I also have this problem in Ubuntu, IDE version 1.6.7. See screenshot.

[image]

It is annoying to have the brace disappear in this way, particularly as it is the "important brace" in this case.

In my tests, if the opening brace is in a different column you can see it:

[image]

VaclavSal, 2016-01-17:
I hope I can use this as an supporting argument for early fix. 
I use AutoFormat and if the leading brace is NOT in first column ( tested on loop ())  , AutoFormat will move it there. 

I am sorry, too busy coding and not having time to get my feet wet in Java.

per1234, 2017-07-04:
I can reproduce the issue in Arduino IDE 1.8.1. The issue does not occur in Arduino IDE 1.8.2 (or 1.8.3). This leads me to believe the issue was fixed by https://github.com/arduino/Arduino/pull/6022.

closed 2017-07-04 by per1234