arduino/Arduino #4339: strange/unnecessary commands in compiled code

night-ghost, 2015-12-24:
small function compiled and then decompiled by avr-objdump
see comments

void deepSleep(uint16_t milliseconds) {
    30d6:       ef 92           push    r14
    30d8:       ff 92           push    r15
    30da:       0f 93           push    r16
    30dc:       1f 93           push    r17
    30de:       cf 93           push    r28
    30e0:       df 93           push    r29
    30e2:       00 d0           rcall   .+0             ; 0x30e4 <_Z9deepSleepj+0xe>
    30e4:       00 d0           rcall   .+0             ; 0x30e6 <_Z9deepSleepj+0x10>
    30e6:       cd b7           in      r28, 0x3d       ; 61
    30e8:       de b7           in      r29, 0x3e       ; 62

    millisCounter += milliseconds;
    30ea:       40 91 e7 03     lds     r20, 0x03E7
    30ee:       50 91 e8 03     lds     r21, 0x03E8
    30f2:       60 91 e9 03     lds     r22, 0x03E9
    30f6:       70 91 ea 03     lds     r23, 0x03EA
    30fa:       48 0f           add     r20, r24
    30fc:       59 1f           adc     r21, r25
    30fe:       61 1d           adc     r22, r1
    3100:       71 1d           adc     r23, r1
    3102:       40 93 e7 03     sts     0x03E7, r20
    3106:       50 93 e8 03     sts     0x03E8, r21
    310a:       60 93 e9 03     sts     0x03E9, r22
    310e:       70 93 ea 03     sts     0x03EA, r23
    uint32_t microseconds = milliseconds * 1000L;
    3112:       9c 01           movw    r18, r24
    3114:       a8 ee           ldi     r26, 0xE8       ; 232
    3116:       b3 e0           ldi     r27, 0x03       ; 3
    3118:       0e 94 ad 34     call    0x695a  ; 0x695a <__umulhisi3>
  
    narcoleptic_calibrate();
    311c:       69 83           std     Y+1, r22        ; 0x01
    311e:       7a 83           std     Y+2, r23        ; 0x02
    3120:       8b 83           std     Y+3, r24        ; 0x03
    3122:       9c 83           std     Y+4, r25        ; 0x04
    3124:       0e 94 c7 17     call    0x2f8e  ; 0x2f8e <_Z21narcoleptic_calibratev>
    if(microseconds < watchdogTime_us ) return;
    3128:       20 91 22 01     lds     r18, 0x0122
    312c:       30 91 23 01     lds     r19, 0x0123
    3130:       40 91 24 01     lds     r20, 0x0124
    3134:       50 91 25 01     lds     r21, 0x0125
    3138:       69 81           ldd     r22, Y+1        ; 0x01
    313a:       7a 81           ldd     r23, Y+2        ; 0x02
    313c:       8b 81           ldd     r24, Y+3        ; 0x03
    313e:       9c 81           ldd     r25, Y+4        ; 0x04
    3140:       62 17           cp      r22, r18
    3142:       73 07           cpc     r23, r19
    3144:       84 07           cpc     r24, r20
    3146:       95 07           cpc     r25, r21
    3148:       08 f4           brcc    .+2             ; 0x314c <_Z9deepSleepj+0x76>
    314a:       63 c0           rjmp    .+198           ; 0x3212 <_Z9deepSleepj+0x13c>
    
    uint16_t sleep_periods = (microseconds - watchdogTime_us) / watchdogTime_us;
    314c:       62 1b           sub     r22, r18
    314e:       73 0b           sbc     r23, r19
    3150:       84 0b           sbc     r24, r20
    3152:       95 0b           sbc     r25, r21
    3154:       0e 94 66 34     call    0x68cc  ; 0x68cc <__udivmodsi4>
    3158:       89 01           movw    r16, r18 // r16r17 =>sleep_periods, right?
    while (sleep_periods >= 512) {
// { what's on?  why we add 0, compare result with 0 - and not use this?
    315a:       e1 2c           mov     r14, r1 //r1 always 0
    315c:       f1 2c           mov     r15, r1
    315e:       c7 01           movw    r24, r14
    3160:       80 0f           add     r24, r16
    3162:       91 1f           adc     r25, r17
[... 91 lines cut ...]
    31f8:       80 e0           ldi     r24, 0x00       ; 0
}
    31fa:       0f 90           pop     r0
    31fc:       0f 90           pop     r0
    31fe:       0f 90           pop     r0
    3200:       0f 90           pop     r0
    3202:       df 91           pop     r29
    3204:       cf 91           pop     r28
    3206:       1f 91           pop     r17
    3208:       0f 91           pop     r16
    320a:       ff 90           pop     r15
    320c:       ef 90           pop     r14
    320e:       0c 94 c4 17     jmp     0x2f88  ; 0x2f88 <_Z22narcoleptic_sleep_downh>
}
    3212:       0f 90           pop     r0
    3214:       0f 90           pop     r0
    3216:       0f 90           pop     r0
    3218:       0f 90           pop     r0
    321a:       df 91           pop     r29
    321c:       cf 91           pop     r28
    321e:       1f 91           pop     r17
    3220:       0f 91           pop     r16
    3222:       ff 90           pop     r15
    3224:       ef 90           pop     r14
    3226:       08 95           ret

matthijskooijman, 2015-12-24:
I took a quick glance at the assembly and indeed, it seems that there are some things that could be further optimized. However, Arduino just uses avr-gcc and is not really in a position to work on the compiler optimizations, so you should consider reporting your issues at the gcc bugzilla (best try to isolate separate issues and find smaller examples for each if you do).

What Arduino IDE version are you using? Since 1.5.something, the compiler version was upgraded, which likely improves the optimizations in use. There is also a avr-gcc 5.something version available on some issue. If you feel up to diving in, I can find the issue and provide you with the link (probably needs a bit of fidgetting tom make it run, though).

night-ghost, 2015-12-24:
this is Arduino 1.6.6
Yes, I realized that compiler was upgraded, the previous version did not invent the code :) Now stacks frame for vars used where old version uses registers...

night-ghost, 2015-12-24:
first look to GCC bugzilla https://gcc.gnu.org/bugzilla/buglist.cgi?quicksearch=avr - the impression that there is no hurry to correct errors...

matthijskooijman, 2015-12-24:
> first look to GCC bugzilla https://gcc.gnu.org/bugzilla/buglist.cgi?quicksearch=avr - the impression that there is no hurry to correct errors...

Yeah, that's also my impressions. Especially missed optimizations, as those are often tricky to fix. They would probably welcome patches, of course :-p

night-ghost, 2015-12-25:
bug reported - https://gcc.gnu.org/bugzilla/show_bug.cgi?id=69049  with new details

per1234, 2015-12-25:
Did you test with the new version of avr-gcc to see if the issue still occurs? I believe this is the issue matthijskooijman mentioned that has the links to avr-gcc 5.1.0: https://github.com/arduino/Arduino/issues/660#issuecomment-120433193

night-ghost, 2016-01-03:
sorry but no - I has Linux-x86 but there are versions of GCC only for Linux-64. After holidays I'll try to compile GCC and then wiil try

night-ghost, 2016-01-18:
Version 5.3 tested and seens that it has'nt this bug

NicoHood, 2016-06-01:
I also want to note a strange output I found:
```
ISR(INT0_vect) {
  // ISR
  PORTD |= (1 << 3);
  PORTD |= (1 << 3);
  PORTD |= (1 << 3);
  PORTD |= (1 << 3);
  PORTD |= (1 << 3);
  PORTD |= (1 << 3);
  PORTD |= (1 << 3);
  // TEST
  PORTD &= ~(1 << 3);
}

ISR(INT0_vect) {
 14a:	1f 92       	push	r1
 14c:	0f 92       	push	r0
 14e:	0f b6       	in	r0, 0x3f	; 63
 150:	0f 92       	push	r0
 152:	11 24       	eor	r1, r1
  // ISR
  PORTD |= (1 << 3);
 154:	5b 9a       	sbi	0x0b, 3	; 11
  PORTD |= (1 << 3);
 156:	5b 9a       	sbi	0x0b, 3	; 11
  PORTD |= (1 << 3);
 158:	5b 9a       	sbi	0x0b, 3	; 11
  PORTD |= (1 << 3);
 15a:	5b 9a       	sbi	0x0b, 3	; 11
  PORTD |= (1 << 3);
 15c:	5b 9a       	sbi	0x0b, 3	; 11
  PORTD |= (1 << 3);
 15e:	5b 9a       	sbi	0x0b, 3	; 11
  PORTD |= (1 << 3);
 160:	5b 9a       	sbi	0x0b, 3	; 11
  // TEST
  PORTD &= ~(1 << 3);
 162:	5b 98       	cbi	0x0b, 3	; 11
}
 164:	0f 90       	pop	r0
 166:	0f be       	out	0x3f, r0	; 63
 168:	0f 90       	pop	r0
 16a:	1f 90       	pop	r1
 16c:	18 95       	reti
```

Is this related or are those instructions expected?

matthijskooijman, 2016-06-01:
As I explained on IRC, most of these extra instructions are expected, though not strictly required. It's probably hard to optimize them away, though.