arduino/Arduino #4343: Custom baud rate to serial monitor

JaakkoFagerlund, 2015-12-25:
At the moment we are limited to just a few select baud rates, while the Arduino board itself can do much more than just 230400 baud. I would thus suggest adding an option of defining a custom baud rate or including more, like 250k, 500k and 1M.

PaulStoffregen, 2015-12-25:
Can it really support those speeds?  Have you actually tested?  Did you test with sustained data flow, or only short bursts, or only single chars with lengthy delays between?

Actual testing is important.  If you make blind assumptions based only on the USART documentation, you'll probably conclude the hardware is capable of much faster speeds than the slow AVR processor can actually handle when running the general purpose, interrupt & buffer based code.

JaakkoFagerlund, 2015-12-25:
AVR processor running at 16MHz has no problems sending 2Mbps out, it is usually limited by the buffer or receiving end and how the code has been done. I haven't personally tested it, but I've seen a scope trace of it.

And even if some Arduino board in some configuration can't handle those speeds, why limit the serial monitor? There are countless other chips, boards and projects that the serial monitor is nice to use with and being able to set a custom baud rate shouldn't be an issue. It doesn't take anything away from anyone.

matthijskooijman, 2016-01-17:
One additional limitation is that on Linux, supporting non-standard baud rates is a bit messy, and the JSSC serial library does not seem to support the right API to set custom baud rates on ttyACMx devices (explained here: https://github.com/arduino/Arduino/issues/3351#issuecomment-113873045). I'm not sure about the situation on Windows and OSX, things might be better supported, so it might make sense to allow custom baudrates on those platforms, or perhaps even on Linux too if a proper error can be displayed when a baud rate failed to set. Issue #3351 might contain more info relevant to this issue, but for now I want to keep this issue as the central one for custom baud rate support and close the reset as duplicates.

matthijskooijman, 2016-01-17:
https://github.com/arduino/Arduino/issues/4439#issuecomment-172229442 additionally suggests that perhaps boards or platforms could define the list of baudrates (or, perhaps just defining additional baudrates on top of the standard ones) would be a good addition. It's a bit of a separate feature from properly supporting non-standard baudrates, but certainly related.

csouers, 2016-04-25:
Hey all. I too would like support for custom baud rates in the serial monitor. Without individually testing each rate, why not make a pseudo-hidden settings option to add your own to the list and leave testing to the user? For example, baud rate 7812 (or 7812.5, but the .5 isn't crucial). It works on the devices and in separate serial apps, so I believe it would within the IDE as well.

ohsix, 2017-03-15:
i'm using a cp2102 with an esp8266, it supports (up to) 921600 baud rate and i'd like to use it.
i expected to be able to click on the number in the serial monitor ui to put in an arbitrary number, and the dropdown just providing common values

being able to define them in boards.txt or something would be sufficient, but literally any way to do it without rebuilding the ide would be great

Perlovka, 2017-06-03:
> One additional limitation is that on Linux, supporting non-standard baud rates is a bit messy
I wrote a simple Qt based serial terminal, and there is no problem to set custom baud rate.
So the problem is not Linux, but Java ;)

matthijskooijman, 2017-06-06:
@Perlovka, it's a bit of both: The custom-baudrate-code in Linux is a bit of a mess (as mentioned [here](http://marc.info/?l=linux-serial&m=143049135615695&w=2) and [here](https://github.com/arduino/Arduino/issues/3351#issuecomment-113873045), and the JSSC library only supporting one of the possible ways to set custom baudrates, that doesn't work with the ACM driver (IIRC). There does not seem to be any "recommended" way to set baudrates either, the one that works best (for most devices) was discouraged by kernel developers for being an internal API or something (again IIRC).

owendelong, 2019-04-14:
Regarding the request in #4439, I've tried running the ESP8266 board I have with an FTDI serial adapter at 74880 baud (which is close th toe requested 76800 baud and it seems to be "close enough" to actually work (at least mostly).

Below is the Serial Monitor contents when booting ESP8266 at 160Mhz with Seria lMonitor at 74880 Baud...

`
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
ve3de77f6
~ld

`

ycuk, 2019-07-23:
hey, Authors, just add typical values more, please. Some boards use in core lib 460800, 921600, etc, neither 1000000 nor 2000000. thanks

owendelong, 2019-07-24:
Most multiples of 78600 (up to 460800) are in use somewhere. 9600, 19200, and 38400 are also common. 

I’m not aware of any micro controllers out there that don’t fit into the above categories. 

Owen

owendelong, 2019-07-24:
Yeah, the choices above 230400 for serial port monitor speeds are a bit odd.

I have noticed, however, that they’re also a bit “flexible”. For example, native ESP8266 default is 76800, but there’s no such setting in the Arduino Serial monitor.

However, 74880 works just fine to decode the 76800 bps output from the chip during boot.

I suspect that 500000 is probably close enough for 460800 and 1000000 will probably work with 921600.

Alternatively, dig out the code and submit a pull request.

Owen

aaronw2, 2020-04-12:
I'm doing a lot of work now with the ESP32 and it would be really nice to support 921600 baud which is readily supported in Linux and works fine for programming but not for the serial monitor.

supaplextor, 2020-07-01:
screen /dev/ttyUSB0 921600,n,8,1 as an alternative works on Linux here. Using an ordinary esp32 dev board.

ohsix, 2020-07-01:
well, if that was an actual alternative I wouldn't need arduino to support it, but arduino closes the serial port and does some other things when you program the part, then reopens them automatically

tmrb, 2021-06-11:
To solve the baud rate problem, we can choose the frequency of 921600 MHz in the Platformio IDE.

VeniceInventors, 2021-12-06:
I would second the suggestion that the baud rates >115200 in the serial monitor be multiples of 115200 like 230400, 460800, 921600 and 1843200, or, the last value.could be a user-defined rate from the preferences.
With the ESP32 at 921600, and the monitor at 1000000, the output isn't readable so it's not "close enough".

DurvalMenezes, 2021-12-13:
> the last value.could be a user-defined rate from the preferences

I second that. My ESP8266 firmware outputs its initial messages at 921600 bauds, and I can't read them using Arduino IDE serial monitor, really irritating. Being able to set the baud rate in preferences would really solve this for me and for many others.