arduino/Arduino #4345: App Crashing Right After Launch

viktorievasova, 2015-12-25:
Hi,
I have cloned the repository and tried to run the Arduino application according to this guide:
https://code.google.com/p/arduino/wiki/BuildingArduino

I am working on OS X. The app compiled with no errors, but when I've tried launching it, it immediately crashes with the following exception (see below). Any ideas what's causing this? Are there any pre-requisities I need to install manually? Thanks!

Viktorie

```
Error: A JNI error has occurred, please check your installation and try again
Exception in thread "main" java.lang.NoClassDefFoundError: LibraryList
	at java.lang.Class.getDeclaredMethods0(Native Method)
	at java.lang.Class.privateGetDeclaredMethods(Class.java:2701)
	at java.lang.Class.privateGetMethodRecursive(Class.java:3048)
	at java.lang.Class.getMethod0(Class.java:3018)
	at java.lang.Class.getMethod(Class.java:1784)
	at sun.launcher.LauncherHelper.validateMainClass(LauncherHelper.java:544)
	at sun.launcher.LauncherHelper.checkAndLoadMain(LauncherHelper.java:526)
Caused by: java.lang.ClassNotFoundException: LibraryList
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 7 more
```

viktorievasova, 2015-12-28:
So this is a problem only in the master branch. In ide-1.5.x it works fine.

facchinm, 2016-01-07:
Hi @viktorievasova ,
it looks like the IDE didn't compile completely. The google code instructions are quite outdated, so you could try recompiling with 
```cd build && ant clean build run```
Make sure your JAVA_HOME env variable is pointing to a valid JDK8 installation

sandeepmistry, 2016-03-09:
@viktorievasova please see @facchinm's comment above.

Updated build instructions can also be found [here](https://github.com/arduino/Arduino/wiki/Building-Arduino).

I'm closing this for now, please re-open if you still have issues building.

closed 2016-03-09 by sandeepmistry