arduino/Arduino #4348: installing arduino libraries

ruggb, 2015-12-27:
in this document near the bottom is
Restart the Arduino application. Make sure the new library appears in the Sketch->Import Library menu item of the software. That's it! You've installed a library! 

I believe that should be
Restart the Arduino application. Make sure the new library appears in the Sketch->INCLUDE Library menu item of the software. That's it! You've installed a library!

matthijskooijman, 2015-12-27:
What document exactly? Could you provide a link?

ruggb, 2015-12-27:
Sorry, I thought the title would be sufficient

https://www.arduino.cc/en/Guide/Libraries#toc5
it is p/o manual installation, 5 lines from the bottom

lmihalkovic, 2015-12-27:
IMHO the only thing missing is the app not forcing an exit/restart cycle. I will try yo address that in my fork, ithink you might want to do the same.

per1234, 2017-11-14:
Thanks @ruggb! The obsolete menu path in the documentation reported here has since been fixed. Unfortunately some typos were introduced but that's a topic for a fresh issue.

closed 2017-11-14 by per1234