arduino/Arduino #4350: Improve Ctrl-Tab navigation

lmihalkovic, 2015-12-27:
The comments to #195 contained a small detail that has gone un-noticed. It was suggested that it might warrant its own top level entry.

@Scalpel78 
```
It would be useful if CTRL-TAB switched between the files you last used, and not EVERY file. I 
want to be able to quickly jump between two files, not step through all files in the project.
```

@lmihalkovic
```
[...] have a sub-group targetting only a subset of all the files (anyone knowing a app
modeling this for inspiration? - xcode has something like that with menus, but i have to check
if there are shortcuts for it, or to see when files come out of that list)
```

per1234, 2015-12-27:
In Firefox I use ctrl + 1-8 for this. I think it's best to leave ctrl+tab for its standard usage. Although using ctrl + 1-8(or ctrl + 1-0)does limit the number of tabs you can switch between it would not be an issue if the tabs were rearrangeable.

lmihalkovic, 2015-12-27:
Thx for the suggestion. As i recall intellij has ctrl-1/xx for bookmarks. So it looks like a simple thing to do is to have a *circular buffer* of the latest n entries and then use Ctrl-n to jump directly to the matching file, regardless of where it is located (in my case that also includes jumping to the source file from a library in a different set of tabs - i treat library/sketch as its own set of tabs)

lmihalkovic, 2016-01-07:
so I wrote the code to cycle through file and only jumping when ALT is released. turns out it displaying a summary of the possible target (like the ALT-Tab app switcher) for each arrow pressing and only jumping when the right file is selected works fine and is surprisingly convenient: it means one can cycle through the list of files quickly to find the desired file without being delayed by the inertia of actually switching tab. you may want to look into it more closely too

lmihalkovic, 2016-01-07:
switching tab contents happens when ALT key is released

[image]
[image]

lmihalkovic, 2016-01-19:
Thank you for the feedback

closed 2016-01-19 by lmihalkovic