arduino/Arduino #4353: Reset button to Serial Monitor

night-ghost, 2015-12-28:
Sometimes hardware Reset button unavailable (Arduino via RDP for example) so needs ability to reset without reflash

NicoHood, 2015-12-28:
What a simple feature. Never thought of this, but it makes totally sense. I like the idea.

matthijskooijman, 2015-12-28:
This button should probably be called "Reconnect", though, since boards do not necessarily reset on (re)connect (most though, but not all).

NicoHood, 2015-12-28:
Or we just add buttons for DTR and RTS. Then the user can manually pull those lines, for those who want to hold the line down. I heard that some people need this for ESP anyways. Then its even more functional.

matthijskooijman, 2015-12-28:
That would work (See #3386), though it might be less novice-friendly perhaps.

NicoHood, 2015-12-28:
That is totally different from what the user wants here.
We do not want to set DTR/RTS at the beginning fixed, we want to change it at runtime in the serial monitor, to actually reset the mcu.

matthijskooijman, 2015-12-29:
@NicoHood, that issue also talks about having the pins GUI-configurable, though I now see it's not the original request. Should have read it better before linking :-)

chrstphrchvz, 2016-03-10:
Another user here. One workaround (at least on OS X) is to use baud pulldown menu and select same rate again, which resets the connection, and resets the board if auto-reset is enabled on it.

chrstphrchvz, 2016-03-10:
I think the idea of the "Reconnect" button is helpful in the case where the board was unplugged, then reattached. The serial monitor doesn't automatically reconnect in that case, nor does it acknowledge that the board was unplugged, but a "Reconnect" button is a convenience over reopening the serial monitor.

lpdunwell, 2018-10-31:
great. so in nearly 3 years this has not been implemented?

alexceltare2, 2019-06-11:
> great. so in nearly 3 years this has not been implemented?

I agree, the IDE has plenty of spare "square" buttons besides Verify, Upload, New, Open and Save. Also the Serial monitor can add an extra button for "Reset Target" as well.