arduino/Arduino #4354: Needs ability to skip initialization of timers 1 & 2

night-ghost, 2015-12-28:
Some complex project uses that timers "manually" but init() in wiring.c always initializes them to PWM. Why not to surround code (lines 284-327 in wiring.c) with #if ! defined(SKIP_TIMER1_INIT) / #endif so one can define such "magical" constants and escape from unneeded code. Now I forced to patch some of that files to adopt to my projects

matthijskooijman, 2015-12-28:
Adding a skip constant currently also requires modifying platform.txt or boards.txt to insert the appropriate `-D` option to the compiler commandline, so that won't help directly (there is an issue about making custom options easier, I think).

However, why is the PWM initialization a problem? In your sketch's setup function (or a library function called by it), you can just change the timer configuration - the Arduino code will no longer touch it, unless you do an `analogWrite` for that pin (which you obviously shouldn't if you're using the timer for something else).

There is one timer that's also used for `millis()` and friends, but I think that's timer0.

night-ghost, 2015-12-28:
yes I can re-init in setup() but it uses additional commands that wastes space in tiny flash

no one hurt: all still uses like before, but those who need it can make
#define SKIP_TIMER2_INIT
#include <Arduino.h> (see also https://github.com/arduino/Arduino/issues/4352)

NicoHood, 2015-12-28:
I guess you want something like this, and then the ability to disable the timers:
https://github.com/arduino/arduino-builder/issues/15

I'd also like the arduino system to be more pluggable. We all know that some core features are nice, but not optimal for advanced users. However there are still advanced libraries out there for arduino, so we still want to use the IDE. Therefore we need to disable some of the core features sometimes.