arduino/Arduino #4355: some definition changes to save some bytes

night-ghost, 2015-12-28:
-int analogRead(uint8_t pin)
+unsigned int analogRead(uint8_t pin)

because it's really unsigned

-void attachInterrupt(uint8_t interruptNum, void (*userFunc)(void), int mode) {
+void attachInterrupt(uint8_t interruptNum, void (*userFunc)(void), uint8_t mode) {

because mode has only low values

-int digitalRead(uint8_t pin)
+uint8_t digitalRead(uint8_t pin)

because it returns only 0 or 1

NicoHood, 2015-12-28:
doesnt analogwrite() also take an int, while only 0-255 are accepted anyways?

I'd also adapt the changes suggested, if it does not break anything.

matthijskooijman, 2015-12-28:
@night-ghost, you say this saves some bytes, you mean the code compiles smaller with these changes?

The proposed changes look good to me, perhaps a pull request along with some statistics about sketch sizes with and without the PR applied would be useful?

night-ghost, 2015-12-28:
Yes, transferring of uint8_t takes less commands on 8-bit processor than int, and unneeded sign cause using signed arithmetic in expressions like "4000/analogRead()" which can not be used anywhere else.

night-ghost, 2015-12-28:
analogwrite() on other platforms that has hardware DAC uses 10-12bit int

night-ghost, 2015-12-28:
"some statistics" - size difference depends on frequency of calls. attachInterrupt saves 2 words on each call, digitalRead one word on each call. Where project size becomes larger than the memory size you'll count each byte

matthijskooijman, 2015-12-28:
Thanks, sounds like good savings. Care to prepare a pull-request?

vbextreme, 2015-12-30:
If a function returns a value wastes the same time both a int8_t as a int16_t.
With a int8_t the return value will be put into r24 and r25 reset .
With int16_t will be put into r24 and r25 if necessary.
No waste of memory.
To call a function instead you can save a log , but the memory remains unchanged.

night-ghost, 2015-12-31:
if function used as

uint8_t b=digitalRead()

then you right. But when used as

if(digitalRead()==HIGH) 

then check compiled as

if(digitalRead(pushButton) == HIGH)
 208:             lds     r24, 0x0106
 20c:             std     Y+1, r22        ; 0x01
 20e:             call    0x60a   ; 0x60a <digitalRead>
 212:             ldd     r22, Y+1        ; 0x01
 214:             subi    r24, 0x01       ; 1
 216:             sbci     r25, 0            ; 0

You see unneeded sbci?

vbextreme, 2015-12-31:
digitalRead ret 8 bit
```
 1d0:	81 e0       	ldi	r24, 0x01	; 1
 1d2:	09 f4       	brne	.+2      	; 0x1d6 <lowdigitalRead+0x44>
 1d4:	80 e0       	ldi	r24, 0x00	; 0
 1d6:	df 91       	pop	r29
 1d8:	cf 91       	pop	r28
 1da:	08 95       	ret
```
digitalRead ret 16 bit
```
17e:	81 e0       	ldi	r24, 0x01	; 1
 180:	90 e0       	ldi	r25, 0x00	; 0
 182:	21 f4       	brne	.+8      	; 0x18c <digitalRead+0x4c>
 184:	80 e0       	ldi	r24, 0x00	; 0
 186:	02 c0       	rjmp	.+4      	; 0x18c <digitalRead+0x4c>
 188:	80 e0       	ldi	r24, 0x00	; 0
 18a:	90 e0       	ldi	r25, 0x00	; 0
 18c:	df 91       	pop	r29
 18e:	cf 91       	pop	r28
 190:	08 95       	ret
```

actually I was wrong.

tico-tico, 2016-02-19:
I was going to file the same bug about _digitalRead()_ but found this one. So, guys, can any of you create the pull request? Thanks.

tico-tico, 2016-02-28:
Ok, here's my attempt https://github.com/arduino/Arduino/pull/4625 .
I haven't touched interrupts since there is some work on them in the pull section.