arduino/Arduino #4357: Assembler syntax highlighting

matthijskooijman, 2015-12-28:
The IDE supports compiling .S assembly files, but AFAIU it does not have support for syntax highlighting for these files, which would make working with them less than ideal.

@lmihalkovic, you mentioned you had some code for adding this syntax highlighting. Care to submit that as a pull request?

lmihalkovic, 2016-01-07:
building the base was a 30min job (ARM Cortex is still not fully validated). the problem in your codebase is that your Editor is not ready for this kind of extension. Whether or not I redo the work for your Editor or end up sharing mine is a pending question for the moment.

lmihalkovic, 2016-01-16:
Ever considered that in light of your ongoing work on the Editor and how closely syntax highlighting is to its internals, you might be the person with the most current knowledge on this topic!!

zment4, 2021-08-02:
I came to look for this. Apparently no-one has had any hand on it yet? ...I might look at this later, in case get something working. Been a pet peeve of mine lately to not see any syntax highlighting on .S files. Or to be exact, to see _wrong_ syntax highlighting on .S files. I would mind less if it had none enabled.

per1234, 2021-08-03:
>I would mind less if it had none enabled.

This is the behavior of [Arduino IDE 2.x](https://github.com/arduino/arduino-ide).