arduino/Arduino #4359: Arduino 1.6.7 fails to correctly extract function prototypes from .ino, requires forward declarations

ratkins, 2015-12-28:
My project at [this commit](https://bitbucket.org/ratkins/technocolourdreamcoat2016controllertest/commits/7ba31482d36ee26b7c02cfeffcf6a823da6d270a) successfully compiled and built under Arduino 1.6.5-betasomething, with Teensyduino 1.26. It fails to build under 1.6.7/1.27 without forward declaring functions that are defined in the .ino (below `loop()`, from which they're called.)

[This commit](https://bitbucket.org/ratkins/technocolourdreamcoat2016controllertest/commits/90ffcf10689d2199fc25e94f18a78022a25fb71a) contains changes necessary for it to compile.

sebastius, 2015-12-29:
Ah so that's what it is. Been messing for hours...

matthijskooijman, 2015-12-29:
This issue was moved to arduino/arduino-builder#88

closed 2015-12-29 by matthijskooijman