arduino/Arduino #4360: Put the build temporary files under each project folder

flyhawk001, 2015-12-29:
(transfer post from:   http://forum.arduino.cc/index.php?topic=368398.0)

We know build process will generate the temporary files when build code.

The arduino IDE put these files to Windows's Temp folder.
Example for my case:
  C:\Users\Andy Chen 2014\AppData\Local\Temp\build4436557367508469592.tmp

And, does not clean it when user quit IDE.
After build many different projectS, many XXXX.tmp folder exist in our hard disk....  waste HD space !

Why not create \tmp\ under each project folder ?  (same file level as *.ino file)  ?
User also can manual clean up temporary file if they wish.

By the way, any configurate file for IDE(Ex. perference setting), had better also put them under each project folder and create the other sub-folder to keep them.

In short, "group files by project".

lmihalkovic, 2015-12-29:
Although it might seem like a neat solution, the scheme is not without limitations: the examples are located under the execute's folder, which for most operating systems should be treated as read-only.
In my fork it gets even more problematic as i can now make use of INZ (being able to execute a sketch project while leaving it zipped #4206 ). So following this strategy would leave two code paths inside the Codabase and lead to situations where the user might be confused that the IDE would ignore the preference setting.

matthijskooijman, 2015-12-29:
As @lmihalkovic indicates, your suggestion is not as easy as it sounds, so I expect it will not be implemented like that.

I guess the real problem is that the temp files are not cleaned up on exit? They are intended to be deleted, and on my Linux system, it works as expected. What operating system and IDE version are you using? How are you exiting the Arduino IDE (closing the window, using the quit menu option, killing the IDE, does it crash?)? Can you see it leaving a  build folder every time, or just occasionally (i.e. if you try now, does it also leave a build folder, or do you only notice a few of them left after a while)?

flyhawk001, 2015-12-29:
Thanks both of your reply first.
If so, keep the IDE behavior.

For the temp. file issue, it seems no  such problem from version 1.6.6
My temp. file might be left from older IDE version.
We can skip this temp. file issue now.

matthijskooijman, 2015-12-29:
Great, thanks for reporting back. I'll close this issue now - if it ever pops up again, just leave a comment and we can reopen the issue to investigate.

closed 2015-12-29 by matthijskooijman