arduino/Arduino #4364: wifimanager for esp8266

domonetic, 2015-12-30:
https://github.com/tzapu/WiFiManager.git

ESP8266 WiFi Connection manager with fallback web configuration portal

The configuration portal is of the captive variety, so on various devices it will present the configuration dialogue as soon as you connect to the created access point.

First attempt at a library. Lots more changes and fixes to do. Contributions are welcome.
This works with the ESP8266 Arduino platform with a recent stable release(2.0.0 or newer)

Chris--A, 2016-01-01:
You need to format you repo for the library manager (I assume you want it added).

You'll need at the minimum, a `library.properties` file.
And a `keywords.txt` file would be better.

Please read: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

cmaglie, 2016-01-04:
Yes, as @Chris--A already confirmed, the library is not yet ready to be published, I guess we should wait a "request for add" from the author itself.

closed 2016-01-04 by cmaglie

domonetic, 2016-01-04:
Sorry I was hoping to merge changes but took longer than expected. Now are ready

domonetic, 2016-01-04:
You are welcome. I took the audacity to propose to be included in the Library Manager

tzapu, 2016-01-04:
@cmaglie the changes graciously provided by @domonetic have been merged.
if everything is ok we would appreciate this getting included.

thank you very much

cmaglie, 2016-01-04:
The library has been added, thanks!

tzapu, 2016-01-05:
thank you very much