arduino/Arduino #4366: Arduino IDE does not follow standard 4-step C++ compilation process.

turkycat, 2015-12-30:
**Root problem**: It appears that the Arduino IDE does not follow the [standard 4-step C++ compilation process](http://faculty.cs.niu.edu/~mcmahon/CS241/Notes/compile.html). Instead, a sketch file is read and compiled linearly, from top to bottom in what appears to be one or two steps.

**Disclaimer**: It is probably bad form to have libraries depend on `#define` directives that are added to a sketch, but should work if the Arduino IDE followed the standard 4-step C++ compilation process. With the latest addition of the `WiFi101`, using directives like this seems necessary for WiFi libraries that need to support both the newly-added `WiFi101` and legacy `WiFi` libraries, mainly due to the way the new library was written.

**Pretext**:
I am working on support for WiFi communication with Firmata, which accepts a `Stream` type for communications with the library. To support this, I've previously written a `Stream` implementation that wraps the original `WiFi` library and handles the server and client logic internally. This worked fine until WiFi101 was released and I've attempted to add support for the new library while maintaining legacy support.

I've called the sketch `StandardFirmataWiFi` to match other implementations such as `StandardFirmata` and `StandardFirmataEthernet`.

I've called the `Stream` implementation `WiFiStream`.

**Problem I am trying to solve**: Both `WiFi.h` and `WiFi101.h` define `WiFiClass` type, which causes compiler issues if they are both included. The obvious solution would be to use a preprocessing directive to choose one or the other, but this doesn't work for various reasons with various different methods. I will do my best to describe each scenario below and explain why it doesn't work.

**Scenario 1**:
The API for these two libraries is identical or nearly identical. The simplest solution would be to require a `#define` to be specified in `StandardFirmataWiFi`, which should allow `WiFiStream` (or the sketch itself) to decide which WiFi library to include. If the Arduino IDE followed the standard 4-step C++ compilation process, these directives should be evaluated during the first step, before any compilation occurs in the second step.

What actually happens is that compilation happens in a linear process, from top of the file to the bottom. This can be observed with the following example:

*File: DirectivesTest.h*
```C++
#include <HardwareSerial.h>

class DirectivesTest
{
public:
  DirectivesTest()
  {
    Serial.begin( 9600 );
    while( !Serial );
	
#if defined(DIRECTIVE)
    Serial.println( "Directive defined!" );
#else
    Serial.println( "No directive defined." );
#endif
  }
};
```

*File: Arduino sketch*
```C++
#define DIRECTIVE 1

#include "DirectivesTest.h"

void setup() {
  DirectivesTest dt;
}

void loop() {}
```
**Output:** Directive defined!

Modified Arduino sketch, swapping the order of `#define` and `#include`:
```C++
#include "DirectivesTest.h"

#define DIRECTIVE 1

void setup() {
  DirectivesTest dt;
}

void loop() {}
```
**Output**: No directive defined.

It might seem possible for my WiFi sketch to #define the necessary directives before including the stream file, but there are other issues with this as these WiFi libraries and Firmata are each more complex than the simple example above, and contain a large amount of inter-dependent files in WiFi and also in WiFi101. It appears to me that when a header file is included, such as `Firmata.h`, the IDE attempts to recursively enumerate other files in that file's directory. 

Assume that `WiFiStream.h/cpp` exist in `libraries/Firmata/utility/`
```C++
#include <Firmata.h>
#define WIFI_101

#if defined(WIFI_101)
#include <WiFi101.h>
#include "utility/WiFiStream.h"
WiFiStream stream;
#endif
```
**Compile result**: Fails as `WiFiServer` does not name a type.

`WiFiServer.h` is included from `WiFi101.h`. This implies that `#include <Firmata.h>` is attempting to compile files in /utility/. If `#include <Firmata.h>` is moved *below* the `#ifdef` block, the error is: `File: utility/WiFiStream.h cannot be found`.

I also cannot add an `#ifdef` block to the `WiFiStream` class itself (to include `WiFi101.h`), as again the file cannot be found if Firmata is not first included, and if Firmata is included first, the `#define` is not yet evaluated, this is the root problem again.

**Scenario 2**:
Instead of attempting to automagically swap out the libraries in `WiFiStream.h`, I've instead created two files: `WiFiStream.h`, which uses `WiFi.h`, and `WiFi101Stream.h`, which uses `WiFi101.h`. The only difference between these two files is which WiFi library is included in the header. The theory is that there is no more dependency on any `#define` statement in the library files themselves.

```C++
#include <Firmata.h>

//#define LEGACY_WIFI
#define WIFI_101

#if defined(LEGACY_WIFI)
  #include "utility/WiFiStream.h"
  WiFiStream stream;
#elif defined(WIFI_101)
  #include "utility/WiFi101Stream.h"
  WiFi101Stream stream;
#endif
```
**Compile result**:
```
In file included from C:\Program Files (x86)\Arduino\libraries\WiFi\src\utility/wifi_drv.h:26:0,

                 from C:\Program Files (x86)\Arduino\libraries\WiFi\src\WiFi.cpp:20:

C:\Program Files (x86)\Arduino\libraries\WiFi101\src/WiFiUdp.h:27:2: error: 'SOCKET' does not name a type

  SOCKET _socket;

  ^
```
(note that `SOCKET` is a `sint8` typedef defined in `libraries\WiFi101\src\socket\include\socket.h`)

Again, the problem here is that since `Firmata.h` must be included first, other files appear to compile before the `#ifdef` blocks have a chance to be evaluated. The error here is strange, but what appears to be happening is that the `WiFi101Stream` class is first compiled into an object file (typically step 3 of the C++ compilation process), and the compiler is therefore aware of the existence of a file called `WiFiUdp.h` (contained within libraries/WiFi101/src). Then, I assume it is attempting to create an object file for WiFiStream, comes across a reference to `WiFiUdp.h` at libraries\\__WiFi__\\src\utility/wifi_drv.h line 26, and refers to its previous knowledge of this file as it exists in libraries\\__WiFi101__\\src/WiFiUdp.h

This observation can be confirmed by *deleting* the "WiFi" folder inside /libraries/. The error is then:

```
In file included from C:\Program Files (x86)\Arduino\libraries\Firmata\utility\WiFiStream.cpp:5:0:

C:\Program Files (x86)\Arduino\libraries\Firmata\utility\WiFiStream.h:7:18: fatal error: WiFi.h: No such file or directory

 #include <WiFi.h>

                  ^
```

The compiler should not be attempting to compile WiFiStream at all!

If I restore the WiFi folder to its original location, and instead delete the `WiFiStream.h/cpp` files, compilation completes successfully. Since these files no longer exist, and the only attempt to include the header is within an #ifdef block which will not evaluate to true, the sketch has no issues.

Indeed, the best work around I have been able to use for writing `StandardFirmataWiFi` is to delete the files `WiFiStream.h` and `WiFiStream.cpp`. Since these are never included in the sketch without explicitly uncommenting the line `#define LEGACY_WIFI`, the compiler happily removes those lines. With this work around, I have been able to complete the WiFi sketch enabling support for WiFi101. Unfortunately, I do not believe this code can be released for public consumption until this issue is resolved or a suitable work around can be determined for the time being.

I will happily help to resolve this issue in any way I can, and will accept any suggestions for work arounds.

turkycat, 2015-12-30:
I will add that I am using Arduino IDE version 1.6.7, with the latest code found on GitHub for each library mentioned above.

PaulStoffregen, 2015-12-30:
Are you aware that Arduino is simply using gcc to do the compilation?

If there's something wrong with how gcc is compiling C++ programs, this issue would probably need to be addressed upstream by the gcc project.

vbextreme, 2015-12-30:
Such errors are caused by sort of build files, there will be until Arduino not use a "classic" Makefile.

matthijskooijman, 2015-12-30:
@turkycat, I believe Arduino is behaving as expected here, and the problem is mainly in your understanding of the process. I haven't fully understood everything you wrote, but I'll clarify I few things and then we'll see if that helps enough for you to understand what happens.

As @PaulStoffregen said, Arduino uses gcc for compilation, which used the 4-step compilation process you linked (the first three steps are done for every source file into a .o file, the last step links all .o files together).

Then, realize that every source file (.c, .cpp and for Arduino also .ino, but not .h) in a project is compiled independently. This means that any `#defines` in one source file never appear in another source file, only `#defines` from header files can be shared.

Furthermore, you said:

> It appears to me that when a header file is included, such as Firmata.h, the IDE attempts to recursively enumerate other files in that file's directory. 

Which is exactly how the Arduino library system is meant to work. If you include any file from a library, *all* of its source files will be compiled in the link. There is currently no way to include some but not others (which would be how you'd typically solve this using a Makefile approach). Another approach would be to use `#ifdefs` to select what code to use, but that requires modifying the library to select an alternative. There is currently no way (AFAIK) that you can let a `#define` propagate from the sketch into a library (there has been discussion about adding options for this, but nothing was added so far).

As for solutions, I can see two:
 - Create two libraries and let the sketch select one through its includes. You might need a third library for the common code.
 - Offer both alternatives in the same library under different names (e.g. different class names) and let the sketch choose by using the right class name. This might not work properly, since it will probably require having both the wifi and wifi101 library installed (assuming you depend on both in this scenario). If that's ok, this is probably the best bet.

And of course, requiring to modify the library to select the right `#define` in a .h file, which could be the easiest solution (but I believe you tried and didn't get this to work?).

Hopefully you'll be able to figure out what is happening with your testcases with this info. Feel free to ask for clarification if things are still unclear (though, technically, this is not really the goal of this issue tracker...).

turkycat, 2015-12-30:
@matthijskooijman This does help to understand some of what I have been seeing, thank you.

I don't intend to bloat the issue tracker for user error, but the one thing I still can't make sense of is the fact that the legacy WiFi library's `wifi_drv.h` (included from `WiFi.cpp`) includes `WiFiUdp.h`, which exists in its parent directory, and yet somehow it makes the jump to instead use the WiFi101 version of `WiFiUdp.h` located in a completely different library. This is using the second example scenario in my original post, which I believe is following your second suggestion to use completely different class names for the two different stream classes. These never cross reference each other's libraries, and so `WiFi.cpp` should never have knowledge of the existence of `WiFi101` or any of it's source or header files.

At any rate, it probably still won't work with the all of the wrinkles ironed out, as invariably the two WiFi libraries create the same class and therefore object file for `WiFiClass`. I have, at one point, gotten it to the point where the compiler barfs that `WiFiClass` was already defined. It sounds like I need to rethink my approach and likely take the route of using two separate libraries.

matthijskooijman, 2015-12-31:
Right, so the missing piece of the puzzle is that, when you include a .h file, in addition to compiling and linking all of the library's source files, it also puts the library directory on the include path. Additionally, there is only one include path, used both when compiling the sketch, as well as libraries. So when `wifi_drv.h` does `#include "WifiUdp.h", it first searches the current directory (which doesn't have the file) and then the include path. I suppose that the WiFi101 library happens to come before the WiFi library in the include path, so the wrong one is included.

This is actually a bug in the WiFi library. When including files within the same library, it is not reliable to rely on the include path, relative includes should be used instead. In this particular case, it should be `#include "../WifiUdp.h"` to explicitely point at the right file. Perhaps you could have a look around the library to see if there are more cases like these and prepare a pullrequest to fix them?

This might not be sufficient to fix the problem, though. If, for example, in your sketch you also need to include ` WifiUdp.h`, you have no way of telling which of the two libraries you actually want - it also depends on the include path. In this case, hopefully, the `Wifi.h` and `Wifi101.h` (or however they're called) will already include everything you need from each library (using relative includes to get the right files).

matthijskooijman, 2015-12-31:
> At any rate, it probably still won't work with the all of the wrinkles ironed out, as invariably the two WiFi libraries create the same class and therefore object file for WiFiClass. I have, at one [...]

If they have more conflicts, in addition to duplicate header file names, then that will also cause problems indeed. One way to solve this in C++ would be to use namespacing, but this isn't used in the Arduino environment currently, so it is probably not helpful here.

turkycat, 2015-12-31:
I've resolved this by moving the entire implementation of both classes to their .h files.

IMHO, this isn't a serious issue like I originally thought, but I think there is room for improvement here. Due to the way that the libraries are enumerated and automatically recursively compiled (as @matthijskooijman confirmed) it leaves little control for library developers. This probably wouldn't be an issue if depreciated libraries were replaced by new ones that defined unique class and file names. I think my issue was compounded severely by the combination of these circumstances. My solution is a bit of a hack and goes against the philosophy and best practices of using separate class definitions and implementations.

I do appreciate the quick responses and all the info that was given. I'm closing this issue.

closed 2015-12-31 by turkycat