arduino/Arduino #4368: Unable to program AVR chips via Atmel ICE with IDE 1.6.7

DrewOJensen, 2015-12-31:
I've been unable to successfully program/burn a boot loader through the IDE (1.6.7) using an Atmel ICE. I've got the cable properly connected to the AVR output on the debugger/programmer and connect properly to ISP. The MCU is an Atmega328P-AU.

The main issue is as soon as I select the Atmel-ICE for the programmer and I select the Arduino Pro, due to using the same MCU and general design (my own PCB), I get the following error when I try to program. "Please select a programmer from Tools->Programmer menu". I have verified this error in both OS X & Windows 10. Both OS's recognize I have the Atmel ICE connected via the USB port. Also, if using a custom SAMD21 board, I can upload sketches when I have the cable plugged into the SAM port on the ICE unit.

Does the Arduino IDE actually support the Atmel ICE programmer?

per1234, 2016-01-01:
Just to avoid duplicate effort I thought I should mention that this issue was also posted to the Arduino forum where I've attempted to provide an answer. http://forum.arduino.cc/index.php?topic=368886

gh-megabit, 2016-01-04:
Note that avrdude 6.0.1 is the current version and this does not support the Atmel-ICE. As such the Atmel ICE can not be used to program or burn bootloaders to AVR chips with the IDE.

You can try to use avrdude 6.1 or 6.2 or use Atmel Studio or some of the tools provided with it.

See https://github.com/arduino/toolchain-avr/issues/17 for more information

facchinm, 2016-01-04:
Thanks everyone, as reported on  arduino/toolchain-avr#17 avrdude 6.1 works great with atmel-ice. The menu entry needs to be created for AVR platform (the one you are referring to is bundled with SAMD (Zero) core.

jockm, 2016-03-14:
As far as I am concerned this never should have made it into the release if it doesn't work.  How is this not an embarrassment?

Simon72, 2017-01-19:
Hello everyone,
I installed the latest version of the Arduino IDE 1.8.0 but even if I connect the Atmel -Ice is never available in the list programmers.
Could you tell me what to do and which files change?

Thank you

facchinm, 2017-01-19:
Tada! #5880

xenomorf, 2017-03-08:
Tada! is nice, knowing that things are moving. However, I just upgraded to arduino ide 1.8.2. And guess what? Still no atmel-ice in the list! Did I overlook something? What can I do next? 
Regards

facchinm, 2017-03-08:
@xenomorf I bet you overlooked in fact :slightly_smiling_face: You can find it under `Programmers/Atmwel-ICE (AVR)`. If it's not there please report here the IDE version you are running and your OS, thanks!

xenomorf, 2017-03-08:
Hi,
I just upgraded to arduino ide 1.8.2, and I’m using MacOS sierra. Here is in attach my programmers.txt file in attach. About half a year ago, I introduced it manually, and then it appeared in the list. However, after the following upgrade, it was gone from the list, but still in the .txt file. It looks like it wasn’t touched by the recent upgrade of 1.8.2 (nightly build). So, what do you think I’m in? Shit…? 
Kind regards and thanks for replying so quickly,
Francisco Ramirez

> 
> @xenomorf <https://github.com/xenomorf> I bet you overlooked in fact 🙂 You can find it under Programmers/Atmwel-ICE (AVR). If it's not there please report here the IDE version you are running and your OS, thanks!

Francisco Ramirez
tel: 0474539272
Palepelstraat 30
9830 Sint-Martens-Latem

atmelice_isp.name=ATMEL-ICE (ISP)
atmelice_isp.program.tool=avrdude
atmelice_isp.program.extra_params=
atmelice_isp.communication=usb
atmelice_isp.protocol=atmelice_isp

avrisp.name=AVR ISP
avrisp.communication=serial
avrisp.protocol=stk500v1
avrisp.program.protocol=stk500v1
avrisp.program.tool=avrdude
avrisp.program.extra_params=-P{serial.port}

avrispmkii.name=AVRISP mkII
avrispmkii.communication=usb
avrispmkii.protocol=stk500v2
avrispmkii.program.protocol=stk500v2
avrispmkii.program.tool=avrdude
avrispmkii.program.extra_params=-Pusb

usbtinyisp.name=USBtinyISP
usbtinyisp.protocol=usbtiny
usbtinyisp.program.tool=avrdude
usbtinyisp.program.extra_params=

arduinoisp.name=ArduinoISP
arduinoisp.protocol=arduinoisp
arduinoisp.program.tool=avrdude
arduinoisp.program.extra_params=

arduinoisporg.name=ArduinoISP.org
arduinoisporg.protocol=arduinoisporg
arduinoisporg.program.tool=avrdude
arduinoisporg.program.extra_params=

usbasp.name=USBasp
usbasp.communication=usb
usbasp.protocol=usbasp
usbasp.program.protocol=usbasp
usbasp.program.tool=avrdude
usbasp.program.extra_params=-Pusb

parallel.name=Parallel Programmer
parallel.protocol=dapa
parallel.force=true
# parallel.delay=200
parallel.program.tool=avrdude
parallel.program.extra_params=-F

arduinoasisp.name=Arduino as ISP
arduinoasisp.communication=serial
arduinoasisp.protocol=stk500v1
arduinoasisp.speed=19200
arduinoasisp.program.protocol=stk500v1
arduinoasisp.program.speed=19200
arduinoasisp.program.tool=avrdude
arduinoasisp.program.extra_params=-P{serial.port} -b{program.speed}

usbGemma.name=Arduino Gemma
usbGemma.protocol=arduinogemma
usbGemma.program.tool=avrdude
usbGemma.program.extra_params=
usbGemma.config.path={runtime.platform.path}/bootloaders/gemma/avrdude.conf

# STK500 firmware version v1 and v2 use different serial protocols.
# Using the 'stk500' protocol tells avrdude to try and autodetect the
# firmware version. If this leads to problems, we might need to add
# stk500v1 and stk500v2 entries to allow explicitely selecting the
# firmware version.
stk500.name=Atmel STK500 development board
stk500.communication=serial
stk500.protocol=stk500
stk500.program.protocol=stk500
stk500.program.tool=avrdude
stk500.program.extra_params=-P{serial.port}

## Notes about Dangerous Prototypes Bus Pirate as ISP
## Bus Pirate V3 need Firmware v5.10 or later
## Bus Pirate V4 need Firmware v6.3-r2151 or later
## Could happen that BP does not have enough current to power an Arduino board 
## through the ICSP connector. In this case disconnect the +Vcc from ICSP connector
## and power Arduino board in the normal way.
buspirate.name=BusPirate as ISP
buspirate.communication=serial
buspirate.protocol=buspirate
buspirate.program.protocol=buspirate
buspirate.program.tool=avrdude
buspirate.program.extra_params=-P{serial.port}

facchinm, 2017-03-08:
I think I'm getting it. I just checked the content of 1.8.2 nightly  and the `programmers.txt` reads
```
avrisp.name=AVR ISP
avrisp.communication=serial
avrisp.protocol=stk500v1
avrisp.program.protocol=stk500v1
avrisp.program.tool=avrdude
avrisp.program.extra_params=-P{serial.port}

avrispmkii.name=AVRISP mkII
avrispmkii.communication=usb
avrispmkii.protocol=stk500v2
avrispmkii.program.protocol=stk500v2
avrispmkii.program.tool=avrdude
avrispmkii.program.extra_params=-Pusb

usbtinyisp.name=USBtinyISP
usbtinyisp.protocol=usbtiny
usbtinyisp.program.tool=avrdude
usbtinyisp.program.extra_params=

arduinoisp.name=ArduinoISP
arduinoisp.protocol=arduinoisp
arduinoisp.program.tool=avrdude
arduinoisp.program.extra_params=

arduinoisporg.name=ArduinoISP.org
arduinoisporg.protocol=arduinoisporg
arduinoisporg.program.tool=avrdude
arduinoisporg.program.extra_params=

usbasp.name=USBasp
usbasp.communication=usb
usbasp.protocol=usbasp
usbasp.program.protocol=usbasp
usbasp.program.tool=avrdude
usbasp.program.extra_params=-Pusb

parallel.name=Parallel Programmer
parallel.protocol=dapa
parallel.force=true
# parallel.delay=200
parallel.program.tool=avrdude
parallel.program.extra_params=-F

arduinoasisp.name=Arduino as ISP
arduinoasisp.communication=serial
arduinoasisp.protocol=stk500v1
arduinoasisp.speed=19200
arduinoasisp.program.protocol=stk500v1
arduinoasisp.program.speed=19200
arduinoasisp.program.tool=avrdude
arduinoasisp.program.extra_params=-P{serial.port} -b{program.speed}

usbGemma.name=Arduino Gemma
usbGemma.protocol=arduinogemma
usbGemma.program.tool=avrdude
usbGemma.program.extra_params=
usbGemma.config.path={runtime.platform.path}/bootloaders/gemma/avrdude.conf

## Notes about Dangerous Prototypes Bus Pirate as ISP
## Bus Pirate V3 need Firmware v5.10 or later
## Bus Pirate V4 need Firmware v6.3-r2151 or later
## Could happen that BP does not have enough current to power an Arduino board 
## through the ICSP connector. In this case disconnect the +Vcc from ICSP connector
## and power Arduino board in the normal way.
buspirate.name=BusPirate as ISP
buspirate.communication=serial
buspirate.protocol=buspirate
buspirate.program.protocol=buspirate
buspirate.program.tool=avrdude
buspirate.program.extra_params=-P{serial.port}

# STK500 firmware version v1 and v2 use different serial protocols.
# Using the 'stk500' protocol tells avrdude to try and autodetect the
# firmware version. If this leads to problems, we might need to add
# stk500v1 and stk500v2 entries to allow explicitely selecting the
# firmware version.
stk500.name=Atmel STK500 development board
stk500.communication=serial
stk500.protocol=stk500
stk500.program.protocol=stk500
stk500.program.tool=avrdude
stk500.program.extra_params=-P{serial.port}

jtag3isp.name=Atmel JTAGICE3 (ISP mode)
jtag3isp.communication=usb
jtag3isp.protocol=jtag3isp
jtag3isp.program.protocol=jtag3isp
jtag3isp.program.tool=avrdude
jtag3isp.program.extra_params=

jtag3.name=Atmel JTAGICE3 (JTAG mode)
jtag3.communication=usb
jtag3.protocol=jtag3
jtag3.program.protocol=jtag3
jtag3.program.tool=avrdude
# Set a bitclock of 0.1us (the fastest supported value). This should
# work regardless of the crystal used, since JTAG doesn't use the MCU
# clock but dictates its own clock.
jtag3.program.extra_params=-B0.1

atmel_ice.name=Atmel-ICE (AVR)
atmel_ice.communication=usb
atmel_ice.protocol=atmelice_isp
atmel_ice.program.protocol=atmelice_isp
atmel_ice.program.tool=avrdude
atmel_ice.program.extra_params=-Pusb
```

So your txt is not matching at all. Could you open `Board Manager` and uninstall the AVR core (if possible)? Maybe the bundle one gets overwritten by the board manager one.

xenomorf, 2017-03-08:
Thanks for assistance, but at the board manager’s (see bottom of mail) core «  Arduino AVR boards » says : INSTALLED. And I don’t see a way to uninstall… clicking around, banging, bad language , … nothing helps. Of course I can still select a previous version, but probably one without ATMEL-ICE stuff…
Kind regards,

Arduino AVR Boards by Arduino version 1.6.17 INSTALLED
Boards included in this package:
Arduino Yún, Arduino/Genuino Uno, Arduino Uno WiFi, Arduino Diecimila, Arduino Nano, Arduino/Genuino Mega, Arduino MegaADK, Arduino Leonardo, Arduino Leonardo Ethernet, Arduino/Genuino Micro, Arduino Esplora, Arduino Mini, Arduino Ethernet, Arduino Fio, Arduino BT, Arduino LilyPadUS     

> I think I'm getting it. I just checked the content of 1.8.2 nightly and the programmers.txt reads
>
> avrisp.name=AVR ISP
> avrisp.communication=serial
> avrisp.protocol=stk500v1 [...]

Francisco Ramirez
tel: 0474539272
Palepelstraat 30
9830 Sint-Martens-Latem

facchinm, 2017-03-09:
This is strange, because the expected output using the nighlty is
```
Arduino AVR Boards Built-in by Arduino version 1.6.18 INSTALLED
Boards included in this package:
Arduino Yún, Arduino/Genuino Uno, Arduino Uno WiFi,...
```
Note the **Built-in** extra string which indicates that the IDE is using the nighly version for the AVR core too. Anyway, nevermind, we'll release the updated IDE soon so the core will be officially released and any problem will be gone :smile:

xenomorf, 2017-03-09:
Dear Martino,
I just installed a new « hourly build » and this time I could update the AVR core. I just did that, and it worked. The ATMEL-ICE is mentioned in the list of programmers.
Thanks a lot for your help,

> This is strange, because the expected output using the nighlty is
>
> Arduino AVR Boards Built-in by Arduino version 1.6.18 INSTALLED
> Boards included in this package:
> Arduino Yún, Arduino/Genuino Uno, [...]

Francisco Ramirez
tel: 0474539272
Palepelstraat 30
9830 Sint-Martens-Latem

closed 2017-03-09 by facchinm

duronebis, 2017-04-05:
I can see it in the list of programmers, but I get this error when trying to upload the sketch:

Arduino:1.8.2 (Windows 10), Scheda:"Arduino Due (Programming Port)"

Lo sketch usa 22124 byte (4%) dello spazio disponibile per i programmi. Il massimo è 524288 byte.
processing.app.debug.RunnerException: Seleziona un programmatore tramite il menù "Strumenti > Programmatore"
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:317)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:89)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:713)
	at processing.app.SketchController.exportApplet(SketchController.java:686)
	at processing.app.Editor$DefaultExportAppHandler.run(Editor.java:2232)
	at java.lang.Thread.run(Thread.java:745)
Seleziona un programmatore tramite il menù "Strumenti > Programmatore"

Questo report potrebbe essere più ricco di informazioni abilitando l'opzione
"Mostra un output dettagliato durante la compilazione"
in "File -> Impostazioni"

facchinm, 2017-04-05:
@duronebis you are selection "Arduino Due" as board, while you should select an AVR board and select ATMEL-ICE (AVR) as programmer

duronebis, 2017-04-05:
So you say that Atmel-ICE in Arduino Ide doesn't work with Arduino Due (Sam) as it does in Atmel Studio?
Must I wait for a ATMEL-ICE (SAM) entry?

facchinm, 2017-04-05:
The SAM core doesn't support any programmer at the moment (see https://github.com/arduino/ArduinoCore-sam/blob/master/programmers.txt ) but feel free to add a Pull request to support Atmel ICE on [SAM repository](https://github.com/arduino/ArduinoCore-sam)

gojimmypi, 2018-03-13:
In order to get the Arduino IDE to successfully program an AVR (in my case ATmega328p), I needed to use the install-filter-win.exe on all 3 drivers: input device, data gateway, and composite with hex Id's: 03EB and 2141 as described here:

https://www.avrfreaks.net/comment/1740521#comment-1740521

Arduini IDE 1.8.5 and Board Manage: Arduino AVR Boards 1.6.21

Simon72, 2018-05-07:
Hi Martino,
I do not know if you can help me but I can not use the ICE to reset my Mega's booloaders.

I use the latest IDE 1.8.6 and I do not see where I can update the programmers.
in the list appears Atmel-ICE but then can not use it.

facchinm, 2018-05-07:
@Simon72 , you need to select `Tools/Programmer` -> `Atmel ICE (AVR)` , then under Board select the Mega 2560 and then press `Burn bootloader`.
If something goes wrong, paste here the **verbose** output ,thanks

Simon72, 2018-05-07:
Grazie Martino!
Here's what the avrdude sends me:
`Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega2560 -cjtag3isp -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xD8:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : jtag3isp
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2110, 0x2140

avrdude done.  Thank you.

Error while burning bootloader.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
Verifying the card driver, however, I find that the VID 0x03eb and PID 0x2140 are correct.

matthijskooijman, 2018-05-07:
Doesn't this need libusb drivers installed on Windows? And some driver selection thingamabob to make sure that libusb can claim the device, rather than some other driver? (The latter might only be the case when Atmel Studio has been installed at some point, which also installs drivers for these devices)

facchinm, 2018-05-08:
@Simon72 the output should read something like:

```
... avrdude.conf -v -patmega2560 -catmelice_isp -Pusb -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xD8:m -Ulfuse:w:0xFF:m 
```
Are you sure you selected `Atmel-ICE (AVR)` under Programmer menu?

Simon72, 2018-05-08:
Sorry Martino I thought it was the same but here is the result with Atmel-ICE AVR:
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega2560 -catmelice_isp -Pusb -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xD8:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : usb
         Using ProgrammeError while burning bootloader.
r              : atmelice_isp
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2141

avrdude done.  Thank you.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Simon72, 2018-05-08:
[image]

Simon72, 2018-05-08:
This is the my print screen:
[image]

Simon72, 2018-05-08:
This is my print screen:
[image]

Simon72, 2018-05-08:
Hi Martino,
With Zadig I find myself two Atmal-ICE devices, which do you think I have to replace the driver for?
[image]

matthijskooijman, 2018-05-08:
@Simon72, you probably need to change it for both.

Simon72, 2018-05-08:
@matthijskooijman Thanks Matthijs,
Did you then use the filter to use the ICE also with AS7?
I would like to keep using my PC on both environments.

matthijskooijman, 2018-05-08:
@Simon72 AFAIU you must use Zadig to switch the driver whenever you switch from the Arduino IDE / avrdude to Atmel Studio and back.

Simon72, 2018-05-09:
@matthijskooijman On [AVRfreak](https://www.avrfreaks.net/forum/atmel-ice-avrdude-cannot-find-device#comment-2462626) it's speak of filter libusb for automatic switching to different editors

matthijskooijman, 2018-05-09:
@Simon72, I don't see where they talk about automatic switching there. I think they talk about libusb filtering of devices, which (I think) only means that other drivers are prevented from talking to the devices, so libusb can claim them when needed. They do suggest that there are more USB devices to be filtered, but since these extra devices do not show up in your list, I suspect Zadig already handles this (they use a different libusb filter program, which offer more fine grained control, but I don't hink you need that).

Simon72, 2018-05-09:
@matthijskooijman i think that you have reason, I test Zadig

Simon72, 2018-05-10:
I managed with Zadig to select a driver that allowed Avrdude to place the bootloader, the problem is to go back ... now AS7 fails to use the ICE
Zadig does not do the reverse and Windows does not restore the right one ... what a mess!

matthijskooijman, 2018-05-10:
Seems you can revert using the device manager: https://github.com/pbatard/libwdi/wiki/FAQ#help-zadig-replaced-the-driver-for-the-wrong-device-how-do-i-restore-it